nasa-jpl/open-source-rover #133: Failure of ~/catkin_ws$ roslaunch src/osr bringup/launch/osr.launch

JHPHELAN, 2019-06-21:
~/catkin_ws$ roslaunch src/osr bringup/launch/osr.launch
...long list of launch commands, followed by:
...
ROS_MASTER_URI=http://ubiquityrobot.local:11311

ERROR: cannot launch node of type [osr/motor_controller.py]: can't locate node [motor_controller.py] in package [osr]
ERROR: cannot launch node of type [osr/joystick.py]: can't locate node [joystick.py] in package [osr]
ERROR: cannot launch node of type [osr/rover.py]: can't locate node [rover.py] in package [osr]
process[joy_node-4]: started with pid [3103]
[ERROR] [1561070410.392290674]: Couldn't open joystick /dev/input/js0. Will retry every second.

[^C entered here]

^C[joy_node-4] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

A search for the missing .py files shows them to be present but perhaps not in the expected directories? --
	~/catkin_ws$ find ~/ -name motor_controller.py
	/home/ubuntu/catkin_ws/src/osr/scripts/motor_controller.py
	/home/ubuntu/osr_ws/src/osr-rover-code/ROS/osr/scripts/motor_controller.py

	~/catkin_ws$ find ~/ -name joystick.py
	/home/ubuntu/catkin_ws/src/osr/scripts/joystick.py
	/home/ubuntu/osr_ws/src/osr-rover-code/ROS/osr/scripts/joystick.py

	~/catkin_ws$ find ~/ -name rover.py
	/home/ubuntu/catkin_ws/src/osr/scripts/rover.py
	/home/ubuntu/osr_ws/src/osr-rover-code/ROS/osr/scripts/rover.py

This was run on the preloaded Ubuntu/ROS image (source avail on request).
When I get time next week I'll try following the Rasbpian / ROS NASA/JPL recipe on a clean SD.

JHP

ericjunkins, 2019-06-21:
I suspect it is because the instructions currently omit the step to make the files executable as well.

Navigate to each of the files and do the following
`sudo chmod +x <filename>`

The following error:
[ERROR] [1561070410.392290674]: Couldn't open joystick /dev/input/js0. Will retry every second.
Makes sense unless you currently have a controller paired to the system and it is registered as /dev/input/js0. As soon as you connect the xbox controller this will go away

ericjunkins, 2019-06-21:
@vssystemluba I notice a few other things that seem a little off. There should be '_' in the package names not spaces. For instance 'osr_bringup' not 'osr bringup'. This is ROS convention to use underscore for package names. The roslaunch command should also be:

`roslaunch osr_bringup osr.launch`

If the catkin_ws is properly sourced then it will find the packages and launch files

JHPHELAN, 2019-06-21:
The '_' are there, my documentation word processor (WordPerfect), or maybe its my font, wants to hide them so I have to edit manually and sometimes I miss some.

There wasn't a command to 'source' the catkin_ws.  Just catkin_make.
(This apparently doesn't matter or is done with the make as it worked.  See below)

Wasn't sure which _ws to chmod, the catkin_ws or the osr_ws so went with catkin_ws.

Changing the launch command from
    ~/catkin_ws$ roslaunch src/osr_bringup/launch/osr.launch
to
    roslaunch osr_bringup osr.launch
was successful! (up to the joystick since I don't have a controller paired.)  
How do you do that, anyway?  That's my next step.

Appreciate the light-speed responses to my "destructive testing" of the code!

JHP

ericjunkins, 2019-06-21:
"sourcing" the workspace should be done as part of the workspace setup as well.

http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment

ericjunkins, 2019-06-21:
As far as pairing the controller of youre using the Xbox controller all you have to do is plug in the wireless receiver and follow steps on pairing controller to that (its just hold one button on each). From there it'll all be done

JHPHELAN, 2019-06-26:
@ericjunkins 
Failure of "catkin_make" in Raspbian/ROS kinetic/OSR environment:

Since I had trouble above and y'all are using Rasbpian instead of Ubuntu, I started from scratch and loaded the latest Raspbian then followed the directions to install ROS then osr-rover code.
All went reasonably well until near the end when I did the "catkin_make":

Run a "catkin make" of the OSR workspace: 
	pi@Fidelity: ~/catkin_ws $ catkin_make

It did a long build then the following errors:
	CMake Warning at /home/pi/ros_catkin_ws/install_isolated/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
	  Could not find a package configuration file provided by "sensor_msgs" with
	  any of the following names:

	    sensor_msgsConfig.cmake
	    sensor_msgs-config.cmake

	  Add the installation prefix of "sensor_msgs" to CMAKE_PREFIX_PATH or set
	  "sensor_msgs_DIR" to a directory containing one of the above files.  If
	  "sensor_msgs" provides a separate development package or SDK, be sure it
	  has been installed.
	Call Stack (most recent call first):
	  osr_msgs/CMakeLists.txt:10 (find_package)

	-- Could not find the required component 'sensor_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
	CMake Error at /home/pi/ros_catkin_ws/install_isolated/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
	  Could not find a package configuration file provided by "sensor_msgs" with
	  any of the following names:

	    sensor_msgsConfig.cmake
	    sensor_msgs-config.cmake

	  Add the installation prefix of "sensor_msgs" to CMAKE_PREFIX_PATH or set
	  "sensor_msgs_DIR" to a directory containing one of the above files.  If
	  "sensor_msgs" provides a separate development package or SDK, be sure it
	  has been installed.
	Call Stack (most recent call first):
	  osr_msgs/CMakeLists.txt:10 (find_package)

	-- Configuring incomplete, errors occurred!
	See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeOutput.log".
	See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeError.log".
	Invoking "cmake" failed		
:
I did a "find" to see if the missing sensor_msgs* were there and at least something exists:

pi@Fidelity:~/catkin_ws $ find ~/ -name sensor_msgs*
	/home/pi/ros_catkin_ws/src/genmsg/test/files/sensor_msgs

I feel like I'm ALMOST there!

JHP

ericjunkins, 2019-06-26:
I'm seeing a little bit of inconsistencies in path stuff here:

~/**catkin_ws** $ catkin_make

/home/pi/**ros_catkin_ws**/install_isolated/share/catkin/cmake/catkinConfig.cmake:76 (find_package):

/home/pi/**ros_catkin_ws**/src/genmsg/test/files/sensor_msgs

Some of these have a "ros_catkin_ws" and some have a "catkin_ws".

Sensor_msgs is a default package of ROS and if it's failing that it is because there is something not set right with the path of your system. can you verify path and folder names for me

JHPHELAN, 2019-06-27:
The directions say go to the "catkin_ws".  Should it be "ros_catkin_ws"?
So tried:
 	pi@Fidelity: ~/ros_catkin_ws $ catkin_make
but got:
	virtual memory exhausted: Cannot allocate memory
        ...
	Invoking "make -j4 -l4" failed
So tried:
	pi@Fidelity: ~/ros_catkin_ws $ catkin_make -j2 
and got:
	...
	[100%] Built target record
So suggest updating directions to "ros_catkin_ws" and "-j2"

JHP

JHPHELAN, 2019-06-27:
... but when I try again (regardless from which directory I try):
$ roslaunch osr_bringup osr.launch 
I still get the same error:

[osr.launch] is neither a launch file in package [osr_bringup] nor is [osr_bring                                                               
up] a launch file name 
The traceback for the exception was written to the log file

If I look for osr.launch I get:

pi@Fidelity:~/ros_catkin_ws $ find ~/ -name osr.launch
/home/pi/catkin_ws/src/osr_bringup/launch/osr.launch

but if I go to that directory I still get the same error.
What am I missing?
JHP

ericjunkins, 2019-06-27:
Truthfully the workspace can be called whatever you want it to be as long as you are consistent with the name. I usually name mine '<project>_ws', for example 'osr_ws'. But for simplicity's sake it should be 'catkin_ws'. I don't think it says anywhere in the docs to put the 'ros_' in there. The '-j' option just specifies how many threads to use for the compilation, it should make no difference in the command working or not. 

The roslaunch command is not specific to the directory, it is looking for that specific launch package in the current ros workspace. For instance in our case:

  - command: roslaunch
  - ros package: osr_bringup
  - launch file: osr.launch

This will work for any directory on the machine because it is looking for the `osr_bringup` package in what is set as the current $ROS_PATH. It then looks for the `osr.launch` launch file in that package. Your error is coming from not having the ros path looking in the correct place. Did you do this above comment? This is what sets when you open a new terminal to have the ros path look at the workspace you set up

> "sourcing" the workspace should be done as part of the workspace setup as well.
> 
> http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment

JHPHELAN, 2019-06-27:
I believe the catkin_ws was created by the ROS install.
Then the ros-catkin-ws was created by the OSR install.
(although it could be vice-versa, I'd have to recheck the instructions)
The -j2 was because of the "virtual memory exhausted: Cannot allocate memory" error.
Looking for ROS_PATH I get:
pi@Fidelity:~ $ ROS_PATH
-bash: ROS_PATH: command not found
pi@Fidelity:~ $ ros_path
-bash: ros_path: command not found
pi@Fidelity:~ $ ROS_path
-bash: ROS_path: command not found
pi@Fidelity:~ $ rospath
-bash: rospath: command not found
pi@Fidelity:~ $ ROSPATH
-bash: ROSPATH: command not found
pi@Fidelity:~ $ ROSpath
-bash: ROSpath: command not found
so somehow its not being created

JHP

ericjunkins, 2019-06-27:
okay let's be explicit step-by-step here and @vssystemluba can update the instructions to coincide with this. Let's start over with creating the workspace. I did these exact steps this morning on a new workspace I created. 

  1. `mkdir -p ~/osr_ws/src`
  2. `cd osr_ws`
  3. `catkin_make`
  4. Copy the packages into the '~/osr_ws/src' folder. To be clear they should be
    - osr_bringup
    - osr_msgs
    - osr
    - led_screen
  5. Add the following to the bottom of the bashrc script: `sudo nano ~/.bashrc`
    - source /opt/ros/kinetic/setup.bash
    - source ~/osr_ws/devel/setup.bash
    - source ~/osr_ws/devel/setup.sh
  6. Open a new terminal. `echo $ROS_PACKAGE_PATH`. It should be the following: 
      /home/< youruser>/osr_ws/src:/opt/ros/kinetic/share

  7. `cd ~/osr_ws/src/osr/scripts`
  8. `chmod +x motor_controller.py robot.py rover.py joystick.py`
  9. `roslaunch osr_bringup osr.launch`

ROS guide on creating catkin_ws:
http://wiki.ros.org/catkin/Tutorials/create_a_workspace
Udemy Tutorial ROS for Beginners (I HIGHLY recommend spending the $11 to take this course and learn ROS, it covers all the basics that you will need):
https://www.udemy.com/ros-for-beginners/

JHPHELAN, 2019-06-28:
In last week's Software Steps.pdf:

1.5 Building the Rover Code to work with your ROS installation
Move to the directory containing the main OSR logic: cd /home/pi/osr/ROS
Create a catkin workspace directory: mkdir -p /home/pi/catkin ws/src
Move to your catkin workspace: cd /home/pi/catkin ws
Run a "catkin make" of the OSR workspace: catkin make
If your catkin make command succeeded, you should now see new build/ and devel/ direc-
tories in /home/pi/catkin ws.

Done and check.  Future versions may change this to osr_ws instead of catkin_ws 
From here on I translate as needed to the original catkin_ws instead of rebuilding in a new osr_ws.

Move to your built catkin workspace: cd /home/pi/catkin ws/
Run the roslaunch command (read more about roslaunch at http://wiki.ros.org/roslaunch):

       roslaunch src/osr bringup/launch/osr.launch

this has been corrected to (but not yet in the current directions) to:

       roslaunch osr_bringup osr.launch

5 Add the following to the bottom of the bashrc script: sudo nano ~/.bashrc
- source /opt/ros/kinetic/setup.bash
- source ~/osr_ws/devel/setup.bash
- source ~/osr_ws/devel/setup.sh

In my .bashrc only the first line was present.  Added the other two.

6 Open a new terminal. echo $ROS_PACKAGE_PATH. It should be the following:
/home/< youruser>/osr_ws/src:/opt/ros/kinetic/share

In mine I get:

       pi@Fidelity:~ $ echo $ROS_PACKAGE_PATH
       /home/pi/catkin_ws/src:/home/pi/ros_catkin_ws/install_isolated/share

7, 8 chmod the *.py files     I found mine in their original directories and chmod'ed them

9 roslaunch osr_bringup osr.launch

My xbox controller is attached (& I think, paired)
Doing this I at least get partial success:

pi@Fidelity:~ $ roslaunch osr_bringup osr.launch
... logging to /home/pi/.ros/log/6a327474-9941-11e9-aea8-b827eb2b674d/roslaunch-Fidelity-2263.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Fidelity:36053/

SUMMARY
========

PARAMETERS
 * /accel: 0,0,0,0
 * /battery_high: 18
 * /battery_low: 11
 * /baud_rate: 115200
 * /enc_max: 1480,1755,1240,1230
 * /enc_min: 550,420,360,325
 * /joy_node/autorepeat_rate: 1.0
 * /joy_node/coalesce_interval: 0.05
 * /mech_dist: 7.254,10.5,10.5,1...
 * /motor_controller_addresses: 128,129,130,131,132
 * /motor_controller_device: /dev/serial0
 * /qpps: 0,0,0,0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    joy_node (joy/joy_node)
    joystick (osr/joystick.py)
    motor_controller (osr/motor_controller.py)
    rover (osr/rover.py)

auto-starting new master
process[master]: started with pid [2273]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6a327474-9941-11e9-aea8-b827eb2b674d
process[rosout-1]: started with pid [2286]
started core service [/rosout]
process[motor_controller-2]: started with pid [2293]
process[joystick-3]: started with pid [2301]
process[rover-4]: started with pid [2303]
ERROR: cannot launch node of type [joy/joy_node]: joy
ROS path [0]=/home/pi/ros_catkin_ws/install_isolated/share/ros
ROS path [1]=/home/pi/catkin_ws/src
ROS path [2]=/home/pi/ros_catkin_ws/install_isolated/share
Traceback (most recent call last):
  File "/home/pi/catkin_ws/src/osr/scripts/rover.py", line 3, in <module>
    from osr_msgs.msg import Joystick, Commands, Encoder
ImportError: No module named osr_msgs.msg

Doing a search for the missing piece:
	pi@Fidelity:~ $ sudo find / -name osr_msgs.msg
returns nothing.
It seems osr_msgs.msg is missing.

JHP

ericjunkins, 2019-06-28:
okay so that's progress!

The roslaunch worked, which means it is finding the osr_bringup package and then the launch file within it. This means that you have the directory sourced correctly. It then appears to launch all the nodes within the osr package as well, which means they are found.

The First error: 

ERROR: cannot launch node of type [joy/joy_node]: joy
ROS path [0]=/home/pi/ros_catkin_ws/install_isolated/share/ros
ROS path [1]=/home/pi/catkin_ws/src
ROS path [2]=/home/pi/ros_catkin_ws/install_isolated/share

_However_ the $ROS_PACKAGE_PATH still seems strange to me.

`/home/pi/catkin_ws/src:/home/pi/ros_catkin_ws/install_isolated/share` 

Is what you said you got. Is your main directory 'catkin_ws' or 'osr_ws'. Just trying to make sure we are on the same agreement of what you named yours. Because it is looking in 'catkin_ws' for sure, as well as still the 'ros_catkin_ws' which could be removed. 

Next, the joy package is a default ROS package, and I notice you are missing in the package path the normal ROS stuff as well, which lives at: 

`/opt/ros/kinetic/share`

Can you check the .bashrc file and make sure things are spelled correctly in that? Also make sure that it exists on your file system:

`ls /opt/ros/kinetic/share` 

Lastly the osr_msgs , do you for sure have the following files at:

/home/pi/< workspace>/src/osr_msgs/msg/Commands.msg
/home/pi/< workspace>/src/osr_msgs/msg/Encoder.msg
/home/pi/< workspace>/src/osr_msgs/msg/Joystick.msg
/home/pi/< workspace>/src/osr_msgs/msg/Status.msg

It seems strange to me that it can find the packages in `<workspace>/src/osr/scripts` and not at `<workspace>/src/osr_msgs/msg/...` and I'm wondering if copying them over there was a typo?

And finally @vssystemluba is going to have to take this over the rest of the way, I will be out of the country for almost a month, so he's going to be doing the heavy lifting of stuff for that time.

JHPHELAN, 2019-06-28:
@ericjunkins 

> Is your main directory 'catkin_ws' or 'osr_ws'.
It's catkin_ws as per Software Steps.  There's no mention of osr_ws until this thread.
In the rewrite, is it going to be osr_ws?  Given this thread I'll assume osr_ws.

I'm pretty sure the ros_catkin_ws came from the ROS install directions referenced in the Software Steps.

I'm going to re-install from scratch, merging these additional instructions, and translate from the generic catkin_ws to the specific osr_ws.  

Additional confusion arises as the Raspberry Pi directions advise using the leaner ros-comm version of ROS while the ROS install directions assume desktop-full and translations have to be done there.  I'll skip the update workspace part of the ROS install as nothing was updated after a clean install.  

I'll continue to document my progress step-by-step and share to assist in the Software Steps rewrite.

JHP

JHPHELAN, 2019-06-29:
[OSR install failure 2019.06.28.pdf](https://github.com/nasa-jpl/open-source-rover/files/3341610/OSR.install.failure.2019.06.28.pdf)

I'm trying to follow the Software Steps.pdf directions as modified by this thread but fail at the catkin_make step.  The old/new instructions vary as to the sequence of the catkin_make step.  Attached is my lengthy step-by-step process with the errors documented and discrepancies underlined.

There are errors because cut/paste sometimes looses the "_" in package names but this is eventually caught and corrected (I hope).

The directions on the linked ROS Install are sometimes ambiguous and have errors.

I got farther doing the ros_comm version than this desktop version.  Which are y'all doing?

I'll have to sleep on this (what little time is left.  It's almost 3:00 am) and try again from the Meetup tomorrow.
Perhaps I'll have some feedback by then as to where I ran off the rails.

JHP

JHPHELAN, 2019-07-11:
@ejunkins @ericjunkins @vssystemluba 
Image file of SD card?
Would it be possible to post an .img file of the OSR SD card to Github so we can get running until the ROS install instructions are clarified?  This is assuming that 1) JPL has a working version of the OSR running on ROS and 2) that the hardware is (essentially) the same as the build instructions.
JHP

ericjunkins, 2019-07-11:
I believe that would be possible to do yeah. It is the identical hardware to what you are using now, so that should be all fine. I think there is a few things we would need to do to be able to post am img, but @vssystemluba could take care of those. 

Additionally I think somewhere else is better, it's going to be at minimum 8gb image, I think it might be 16 or 32gb img based on what SD card we used.  Again @vssystemluba would know better on that than I do

JHPHELAN, 2019-07-14:
@ericjunkins @vssystemluba in the 6/27/19 11:02 post it says to run catkin_make BEFORE copying the OSR files into the osr_catkin_ws.  In Software Steps 1.5 it says to move the files first, then run catkin_make.  
**I'm assuming it should be move before catkin_make, correct?**
Working on merging all this info into a set of install directions.
Any luck on posting an image file?
JHP

ericjunkins, 2019-07-14:
Run the Catkin make before hand just to make sure everything is working and setup for the system previous to copying stuff over is all that is there for, easier to debug with that

vssystemluba, 2019-07-15:
@JHPHELAN sorry for the delay on this.  Things have been very busy and the JPL OSR was out and about doing STEM outreach events!

I built a new RPi with Raspbian Buster and will try and get the OSR code up and running on that this afternoon.  I already hit a couple snags due to updated dependencies, so we'll see how far I get.  If the Buster build fails, I'll re-build with Raspbian Stretch, reproduce the errors that following the documentation produces, and post a workaround both here and in the software documentation. 

Additionally, I've just started a copy of our SD card contents as a backup solution per your suggestion.  I will try and get a link posted by the end of today.  Our OSR SD card happens to be a relatively small 8GB one, but note that the .img will therefore be 8GB.  You should be able to restore it to any SD card that's at least 8GB.

vssystemluba, 2019-07-15:
All right @JHPHELAN... I think I finally figured out why you're having these issues!  Based on my new build on Raspbian Buster (it eventually resulted in the exact same errors you found), here's how to fix the issue:

In short, our custom `osr_msgs` package is failing to build because it has a dependency on the ROS package `sensor_msgs`.  The `sensor_msgs` package is not installed by default with ROS and must be installed manually in order to run.  The commands to install that package are:
```
cd ~/osr_ws
rosinstall_generator sensor_msgs --rosdistro kinetic --deps --wet-only --tar > kinetic-sensor_msgs-wet.rosinstall
wstool init src kinetic-sensor_msgs-wet.rosinstall
```

After installing the `sensor_msgs` package via the commands above, make sure you source your osr ROS environment via:
```
source /opt/ros/kinetic/setup.bash
source ~/osr_ws/devel/setup.bash
source ~/osr_ws/devel/setup.sh
```

(I also place those three lines at the end of my `~/.bashrc` file.  That way, that they are automatically sourced each time I open a new terminal and I never have to re-type them in the future)

From there, you should be able to successfully run `catkin_make`.

Please test these out and let me know what you find.  I'll update the documentation and continue working on posting our working OSR SD card image in the meantime.

vssystemluba, 2019-07-15:
Also, to clear up any confusion about `~/catkin_ws` versus `~/ros_catkin_ws` versus `~/osr_ws`:

- `~/ros_catkin_ws` is the catkin workspace that you create in order to install ROS.  You actually build ROS out of this `~/ros_catkin_ws` workspace.
- `~/osr_ws` is the catkin workspace that you create in order to build the OSR ROS code.  It can be named anything you like, so long as you are consistent everywhere.  We suggest just sticking to `~/osr_ws`.
- `~/catkin_ws` was the name that we originally suggested using for building the OSR ROS code in the software documentation.  After this thread, we realized it was confusing to call it that and we've updated the documentation to use only `~/osr_ws`.

ericjunkins, 2019-07-15:
Good find @vssystemluba! I am fairly confident that we don't use sensor_msgs in that osr_msgs and could be removed as a dependency

JHPHELAN, 2019-07-15:
@vssystemluba @ericjunkins Ah, ha!  Thanks!  I was thinking that there could only be one ?_ws and that it should be named osr_catkin_ws and build everything (including the ROS install) there.  Will update my install plan accordingly and will add sensor_msgs (whether we really need it or not).  Looking forward to revised install directions and the SD card image before I try again from scratch using Buster as that will be the new default Raspbian version.  Should also make for a smooth(-er?) transition to the R Pi 4.
JHP

vssystemluba, 2019-07-15:
Updated install directions have been posted in bb715719cbc01bfc9857d65705ebc4c7dfb569b0 ([Software Steps.pdf](https://github.com/nasa-jpl/open-source-rover/blob/ros-rework/Software/Software%20Steps.pdf)).

I am holding off on posting the full SD card because our rover contains some things that we don't want to post along with the image.  I hoped the cleanse would be trivial, but it'll take a little longer than anticipated for me to cleanse it and upload.  I highly recommend going through the install steps as outlined in the [Software Steps.pdf](https://github.com/nasa-jpl/open-source-rover/blob/ros-rework/Software/Software%20Steps.pdf) document; I believe you should be able to get the rover built from just those directions on Raspbian Stretch.  If you want to go the Raspbian Buster route, you may also find my installation notes at https://www.tapatalk.com/groups/jpl_opensource_rover/default-raspbian-noobs-download-is-version-buster-t120-s10.html helpful.

JHPHELAN, 2019-07-17:
Building from the archive of Raspbian Jessie with ROS Desktop following the latest Software Steps.  
Got to the part in the ROS install:
3.3 Building the catkin Workspace
Once you have completed downloading the packages and have resolved the dependencies, you are ready to build the catkin packages.
Invoke catkin_make_isolated:
	pi@Fidelity:~/ros_catkin_ws $ sudo ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/kinetic
	Traceback (most recent call last):
  File "./src/catkin/bin/catkin_make_isolated", line 12, in <module>
    from catkin.builder import build_workspace_isolated
  File "./src/catkin/bin/../python/catkin/builder.py", line 66, in <module>
    from catkin_pkg.terminal_color import ansi, disable_ANSI_colors, fmt, sanitize
ImportError: No module named terminal_color

The other potential problems the ROS install directions mentioned did not occur.

I'd hoped that the module terminal_color was unnecessary and we could just skip it, however this clearly shows that ROS failed to install at all:

Note: This will install ROS in the equivalent file location to Ubuntu in /opt/ros/kinetic however you can modify this as you wish.
[NO, it didn’t:]
	pi@Fidelity:~ $ source /opt/ros/kinetic/setup.bash
	-bash: /opt/ros/kinetic/setup.bash: No such file or directory
	pi@Fidelity:~ $ cd /opt
	pi@Fidelity:/opt $ ls
	pigpio  vc

I Googled the error message but their answers didn't fix the problem.

JHP

vssystemluba, 2019-07-17:
To my knowledge, we've never tested building ROS in Jessie.  It should be supported with instructions in the installation that you were following.

Perhaps try using Stretch or Buster if you can't figure out the Jessie build?

JHPHELAN, 2019-07-17:
Sorry, I meant Stretch!
JHP

JHPHELAN, 2019-07-17:
@vssystemluba 

> I am holding off on posting the full SD card because our rover contains some things that we don't want to post along with the image.
By making a clean install using your latest directions y'all should be able to post a .zip file a la Raspbian, no?
JHP

JHPHELAN, 2019-07-20:
@ejunkins @ericjunkins @vssystemluba ALMOST THERE!  So, so close!!
Fails with ERROR: cannot launch node of type [joy/joy_node]: joy
Raspbian Buster install - success!
ROS Kinetic install - success!
OSR code install - success!
$ roslaunch osr_bringup osr.launch      good until above error.
BEWARE of cut/paste commands!  Word processors can 
1) change a double hyphen into one long dash and 
2) fail to copy underscores "_" which have caused several errors.
Proofread your commands before [ENTER] and enter by hand when necessary.
The entire saga including some points of confusion is here:
[OSR Software Install ALMOST 2019.07.19.pdf](https://github.com/nasa-jpl/open-source-rover/files/3413177/OSR.Software.Install.ALMOST.2019.07.19.pdf)
Can y'all help me with this last step?!?
Jim Phelan
Mars rover project lead
USAi Labs
(formerly Houston Robotics)

ericjunkins, 2019-07-20:
The joy package is default package in ROS, if it wasn't in your installation bundle you can just install it like any ROS package

JHPHELAN, 2019-07-22:
@ericjunkins @vssystemluba 
"The joy package is default package in ROS, if it wasn't in your installation bundle you can just install it like any ROS package"
Apparently it isn't.  Following the Software Steps in
[https://github.com/nasa-jpl/open-source-rover/blob/ros-rework/Software/Software%20Steps.pdf](url)
and the standard Raspbian Buster and ROS kinetic install steps the launch persists in failing to find joy_node.  It has been reported elsewhere that joy fails to install with ROS kinetic.  Following the ROS instructions for adding packages also failed to install joy.  Although cautioned against mixing source installs with apt-get installs, that too was attempted and failed to find the package.
The updated complete procedure is here: 
[OSR Software Install Plan FAIL 2019.07.21.pdf](https://github.com/nasa-jpl/open-source-rover/files/3415316/OSR.Software.Install.Plan.FAIL.2019.07.21.pdf)

There is some potion or incantation that is missing from this spell.
Does JPL have a working ROS based rover built from a clean install following the published instructions?
If so, please publish the SD card image.  There shouldn't be any superfluous data that would need to be cleaned from a fresh install on a newly formatted card.  Thanks!!

Jim Phelan

ericjunkins, 2019-07-22:
I believe the complication comes from a 'clean install' is that it is done at JPL and needs to connect to networks in order to move the software packages onto it. I'm finally back from vacation now though, so can discuss with @vssystemluba tomorrow and see what we can do.

vssystemluba, 2019-07-25:
@JHPHELAN: https://drive.google.com/open?id=18Tzx2tzmwUHOpBue-Hew3XYpXZIEQpiF

See above for a clean install of Stretch and ROS.  We checked out and built the rover code on it, but as a general practice you should always pull the latest code from GitHub, copy it to your osr_ws, and run the catkin_make steps.

Note: The above file is ~5GB.  Unzipped, the image is 32GB since it came off of a 32GB card.

vssystemluba, 2019-07-26:
@JHPHELAN I'm going to go ahead and marked this as closed.  We now have the following:

- Updated Software Steps.pdf with corrected instructions
- Verified that a clean install of Stretch and following the steps in Software Steps.pdf gets the code running successfully
- Verified that a clean install of Buster and following the steps in Software Steps.pdf (plus the additional steps outlined in ) gets the code running successfully
- Posted a working image of Stretch with ROS installed (see above)

Please let us know of your progress.  If you have any other issues with the installation or Software Steps, let's go ahead and open another issue to deal with each of those specific issues.  Thanks for helping us debug this!

closed 2019-07-26 by vssystemluba

JHPHELAN, 2019-07-27:
@ericjunkins @vssystemluba 
	Get prebuild Jessie+ROS image from:
	https://drive.google.com/open?id=18Tzx2tzmwUHOpBue-Hew3XYpXZIEQpiF
		Extract all says “invalid zip file”
		7-Zip says “Headers Error.  Warning: 32-bit overflow in headers”
		Write .img to 32GB SD card (same problem w/ 128GB SD card)
		after writing to SD card with Win32 Disk Imager
Seem like image file is corrupt.  Do you have SHA256's has of .zip and .img files?
Will try from scratch with the new Software Steps directions
JHP