nasa-jpl/open-source-rover #134: Need updated instructions for battery integration with PCB

ireasor, 2019-07-05:
We're wrapping up our rover build using the monolithic PCB and are not 100% sure about how we should be tying in the battery, voltmeter, and switch.  In the old instructions, we went from the battery to the switch, to the voltmeter, to the system.  On the PCB, I see terminals for BATTERY IN as well as POWER SWITCH, which makes me think that the new system may be a little more sophisticated.  It would be great if the new version of the PCB Assembly doc could include this information.

vssystemluba, 2019-07-08:
I am currently working to incorporate the information you're looking for into the new build documents.  However, here's a quick sketch for folks to use in the meantime if you're stuck:

[image]

ericjunkins, 2019-07-26:
Fixed in da93288bcd125ea098a3b9da72a2fec6f436b69f

closed 2019-07-26 by ericjunkins