nasa-jpl/open-source-rover #136: Add wording to Software Steps document about recommending building ROS from source, not using the pre-built Ubuntu / ROS image

vssystemluba, 2019-07-22:


vssystemluba, 2019-07-22:
(as per the discussion on the OSR forum at https://www.tapatalk.com/groups/jpl_opensource_rover/viewtopic.php?p=792#p792)

vssystemluba, 2019-07-26:
We may also alternatively just recommend downloading a pre-built image that we supply for Stretch / ROS.

vssystemluba, 2019-07-26:
Also add clarity about WHICH version of Raspbian to install, since the installation process is definitely trickier in Buster.

Also add a link to the existing image for those who don't want to / can't build their own image.

JHPHELAN, 2019-07-27:
The Software Steps directions to refer to the ROS.org instructions need clarification.  There are several choices of which version of ROS to install desktop-full, desktop, ros-comm, etc.  Which one should we install?  There were also missing packages, ie joy that need to be installed with the main install.  We need a full walk-through of the ROS install.
JHP

ericjunkins, 2019-07-27:
It's ros desktop full. It has joy package included, it was verified that worked and was included in our install a few days ago

JHPHELAN, 2019-07-28:
pi@Fidelity:~ $ cd ros_catkin_ws/
pi@Fidelity:~/ros_catkin_ws $ rosinstall_generator desktop-full --rosdistro kinetic --deps --wet-only --tar > kinetic-desktop-full-wet.rosinstall
**The following unreleased packages/stacks will be ignored: desktop-full
No packages/stacks left after ignoring unreleased**
JHP
USAi Labs

JHPHELAN, 2019-07-28:
The ROS install refers to:
http://www.ros.org/reps/rep-0131.html#variants which is old ROS groovy variants
the current rep is:
http://www.ros.org/reps/rep-0142.html which is indigo and newer.
apparently, between the two, desktop-full became desktop_full  [hyphen to underscore]!
JHP
USAi Labs

ericjunkins, 2019-07-28:
reasons like that above are why we don't want to spell out individual every step-by-step for the ROS install, and rather link to the maintained ROS install guides. Things are bound to change with their stuff and it's much easier to go to their pages and learn whats wrong

JHPHELAN, 2019-07-28:
We should at least specify "desktop_full", show the correct command to execute it and point out pitfalls in linked directions when we are aware of them.
JHP

ericjunkins, 2019-07-28:
agreed, will update wording

JHPHELAN, 2019-07-30:
Here's another critical bit of info -- ROS changed their GPG keys as of 6/7/19.  The ROS install instructions are not updated and will not work unless you get the new keys.  I think this is why my last attempt to build ROS from source failed.  The OSR Software Steps should point this out.  If we ever get a proven road map we'll share.
JHP
USAi Labs
[https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454](url)

timsent, 2019-08-04:
Leo and I are trying to build it this evening. Thanks for letting us know!

vssystemluba, 2019-08-06:
@JHPHELAN thanks for the follow-up.  I've added a section in the Software Steps doc highlighting the known "gotchas", as well as indicating that you should be installing/building the desktop_full version.

commit d5b312e0 (2019-07-26): Added clarity about Raspbian versions and a link to the pre-built image.  Fixes #136.
commit 1a500ad3 (2019-08-06): Added warnings & gotchas to ROS installation step.  Fixes #136.
closed 2019-08-06 by vssystemluba