nasa-jpl/open-source-rover #137: Incorrect Instructions in Software Steps.pdf

camm73, 2019-07-25:
In section 1.5, the very last command currently says `sudo nano /.bashrc`, however this is not in the proper directory. Should be changed to `sudo nano ~/.bashrc`.

Also before moving the the files in the from the cloned repo to the "orc_ws" workspace in section 1.5, the python files must be made executable, or else they will cause an error. This can be done be running `cd /home/pi/orc/ROS/scripts` and then running `chmod +x <insert python script filename>` for each of the python scripts.

JHPHELAN, 2019-07-25:
@camm73 You are correct!  Note typo - "orc_ws" should be "osr_ws"
JHP

ericjunkins, 2019-07-25:
I don't know I rather like the idea of the orc workspace

[image]

commit b228eb23 (2019-07-26): Final edits of Software Steps.pdf.  Closes #137.
closed 2019-07-26 by vssystemluba