nasa-jpl/open-source-rover #142: is the orientation of the Dual Side Mounts step 2.1.1 of the Chassis build wrong?

toebes, 2019-08-02:
In auditing the 3D model, I found when I got to step 2.4.3, the plates didn't match.  Backtracking through the steps, it brought me back to step 2.1.1 which tells to align the sidemounts flat with the top plate.  This corresponds to Figure 3 in step 2.1.2 which definitely shows the tops of the side plates align with the top of the top plate.   However, if you look at Figure 4, you can see that the Dual Side Mounts have rotated and the side plates now abut to the bottom of the top plate.  If I use the orientation from Steps 2.1.1 the front and back plates don't align.   However if I rotate them to match figure 4, everything works out.

billydelectronics, 2019-08-03:
@toebes check out the thread at:  https://r.tapatalk.com/shareLink?url=https://www.tapatalk.com/groups/jpl_opensource_rover/body-assembly-orientation-of-dual-side-mounts-on-t-t97.html&share_tid=97&share_fid=2096986&share_type=t

toebes, 2019-08-03:
@billydelectronics Ah, I missed that thread.  I got about 20 steps in the model before I discovered the issue.  Fortunately it was a lot easier for me to change in the model than it is with the real parts

vssystemluba, 2019-08-06:
@toebes you are very keen in that the orientation in the photos in our build documents was incorrect, as a couple others have noticed and pointed out on our forum as @billydelectronics mentioned above.

Thanks to your great Onshape model, I was able to re-create some clearer pictures with the corrected orientation and have updated the build docs accordingly.

Thanks everyone!

closed 2019-08-06 by vssystemluba

toebes, 2019-08-07:
I'm glad that we have such a collaborative team on this!

commit 98068ce3 (2019-08-06): Fixed orientation of side mounts in Body Build Doc.  Fixes #142.