nasa-jpl/open-source-rover #144: Add Kitspace compatible electronics BOMs?

kasbah, 2019-08-04:
Hey, I noticed in the readme it says:

> The Bill of Materials folder contains (currently just one) Bill of materials file for a specific vendor. We are searching for better ways to help with the ordering process

This is exactly what my open source project [kitspace.org](https://kitspace.org) is for! It lets you buy across the 5 currently supported distributors with a single click. 

To mirror your PCB designs there we need to add a few files: a manifest and the electronics BOMs (can be csv, xlsx or ods) separated into PCB files. When you update the repo on GitHub in the future it will automatically sync your changes. We do have a BOM export script for KiCad as well. 

I would be glad to work with you on this and put this project up. Are you interest?

vssystemluba, 2019-08-06:
@kasbah we're very interested in anything that helps folks get their parts easier, faster, or less expensively!!!

Would you be willing to take our BOM and build out a project on kitspace to show us what that would look like?  It'd be great to get an updated BOM with easier carts / ordering for folks.

Do note that we build our master parts list / BOM dynamically from our build instruction files.  There may need to be a little thought going forwards about maintaining accuracy and consistency, but even as a first pass having something like a kitspace project (even for a static BOM as it exists today) would definitely add value over what we already have available.

I'm going to close out this issue here since it's not an issue against our repository per se, but I'd love to see what you can come up with in kitspace for this project.  We can continue to chat on this issue or on the OSR forum (https://www.tapatalk.com/groups/jpl_opensource_rover/).

vssystemluba, 2019-08-06:
(And FYI if we get a kitspace project / BOM to an acceptable place, we will definitely link to it from our main repository!)

closed 2019-08-06 by vssystemluba

kasbah, 2019-08-07:
Awesome! @kevinb456, would you be up for generating the BOMs and sending the appropriate pull-requests or should I do it? 

I think as a first step we'll add two .csv to the `Bill of Materials Files` folder one for `Arduino_uno_sheild` (sic) and one for `Control Board`. We should be able to use our [KiCad scripts](https://github.com/kitspace/1clickBOM/tree/master/output_scripts/kicad) and then reconcile that with your Digikey BOM. One issue is that you haven't been using the actual PCB schematic references in any BOMs? I am sure we can figure it out though.  

Maybe we can set up a script for you to re-generate these in the future if any of the electronics design change and also incorporate that into your build. 

Interested by your approach in general. In my other job we develop an open source microscope ([openflexure.org](https://openflexure.org)) and we are developing a tool called [git-building](https://gitlab.com/bath_open_instrumentation_group/git-building) to generate BOMs from markdown descriptions. It seems like a very similar approach except you are using `.tex`. I like that your "master parts list" is a spreadsheet (we have been using yaml for a similar purpose but I like this better).