nasa-jpl/open-source-rover #148: Part of step 1.5 (7) of the Mechanical Integration is a duplicate of step 2.2.3 in the Body Assembly

toebes, 2019-08-04:
Given the restructuring of the Body Assembly, it makes sense to eliminate Figure 8 and the first sentence under 7. which says "Attach the clamping hub on the center of the differential pivot to the 0.5 inch aluminum tube above the three washers from the last step". This also makes sense in light of #145 .

vssystemluba, 2019-08-06:
Note: the above also partially duplicate step 2.1 in the Differential Pivot build doc.

vssystemluba, 2019-08-06:
I've refactored the documentation to make the above much clearer by doing the following:

1) Differential Pivot Build Doc now only includes parts for building the assembly directly attached to the differential pivot
2) Body Build Doc now only includes pillow blocks for the body assembly and leaves the differential pivot attachment steps to the mechanical Integration build doc where they belong
3) Mechanical Integration now includes the necessary steps (and parts) for attaching the differential pivot to the body assembly.

The above make the order of operations for that area of the build much clearer and hopefully help others avoid some confusion.

As another interesting note, because of the duplication in steps, we also had a duplication in the parts required, and therefore duplication in the master parts list.  By removing that duplicate step (and thus the duplicate parts), we saved the project about $15 on the build cost.  Good work!

closed 2019-08-06 by vssystemluba

vssystemluba, 2019-08-06:
I just realized that the Body build doc does still contain images that include the clamping assembly that's now only a part of the differential pivot build doc.  I've created #152 to track and fix those images.

commit 4590fbdf (2019-08-06): Removed duplicated steps for attaching diff pivot.  Fixes #148.