nasa-jpl/open-source-rover #149: Broken link to pre-built ROS image

ireasor, 2019-08-04:
In the Software Steps pdf, there is a link to a "Custom JPL image with both Raspbian Stretch and ROS pre-built" that points to https://drive.google.com/open?id=18Tzx2tzmwUHOpBue-Hew3XYpXZIEQpiF.  This URL returns a 404.

ericjunkins, 2019-08-04:
prebuilt image can be found at the following for now: https://drive.google.com/drive/folders/1RbYWDRthpcktqkPEHb7qVqDiy27EiAc1

ireasor, 2019-08-06:
Thanks @ejunkins.  I downloaded the image and have tried to load it onto an SD card with balenaEtcher.  Unfortunately, the image is 32.01GB and the SD card that I have is 31.91GB.  I've tried shrinking the disk image in the MacOS Disk Utility, but I haven't had any luck.  Is there any way on your end to make the image slightly (1-200MB) smaller?  Due to the variation in actual capacity between manufacturers, I expect that others will run into this issue as well.

ericjunkins, 2019-08-06:
Yes we are going to make the image on an 8gb card instead. This requires going through the entire install process on a fresh SD card of that size

ireasor, 2019-08-06:
That makes sense - I imagine that will take some time.  In the meantime, I'll go back to doing it "the hard way".  ;)  Thanks.

commit b7b38174 (2019-08-06): Updated link to custom SD card image.  Closes #149.
closed 2019-08-06 by vssystemluba