nasa-jpl/open-source-rover #152: Update Body Build Doc images to no longer include the differential pivot clamp assembly

vssystemluba, 2019-08-06:


vssystemluba, 2019-08-06:
@toebes I'd love to use photos from the new Onshape model for this if possible.

However, since we moved the duplicated steps related to the differential pivot out of the body build doc, those parts need to be edited out of the Onshape model.  Would you mind clearing the old steps 2.2.2 and 2.3 out of the body build assembly in Onshape and making sure the assemblies are still clean as you intended them to be?  I didn't want to dig too far into the model myself for fear of breaking your beautiful creation.  :)

toebes, 2019-08-07:
I've fixed the model deleting the old steps.  Let me know if you want any additional slicing and dicing of it.

vssystemluba, 2019-08-08:
The model looks good.  I'll update the photos in our build doc based on the new model steps later on today.  :)

Thanks for the rapid update!

commit a7cd2845 (2019-08-09): Updated images in Body Build Doc.  Fixes #152.
closed 2019-08-09 by mikcox