nasa-jpl/open-source-rover #161: Fasteners in Corner Steering Assembly: B5 too long, should be B3. B1 unused.

Roger-random, 2019-08-31:
**RE: Corner Steering Build Doc**

It called for 6-32 3/4" long screws (B5) for fastening corner steering motor mount (S9) through channel (S2) into standoff (T9).

It appears the threaded hole in standoff (T9) is not deep enough to accommodate that length screw. Some report success with 5/8" (B4) but the depth is not consistent. At least one build required going down to 1/2" (B3)

These fasteners (B5) are also used later in the document to fasten pillow bearing blocks (S10) and encoder mount (S31) to standoff (T9). This would consume the 16 (4 per corner) listed in the document.

Possibly related: the parts list (Table 1) in this document also calls for 6-32 1/4" screws (B1) but they do not appear to be used in the assembly.

**Suggestion:**
- In Table 1: replace B1 with B3, change quantity from 24 to 16
- In motor mount instructions: replace B5 with B3

Leave B5 reference and quantity unchanged in Table 1, they are used in pillow bearing block assembly.

Roger-random, 2019-08-31:
Related forum thread: https://www.tapatalk.com/groups/jpl_opensource_rover/screw-length-for-corner-steering-motor-block-throu-t92.html

darpatel, 2020-01-03:
The actually CAD calls for B1s that fixed it for us

Achllle, 2021-01-24:
This has been resolved!

closed 2021-01-24 by Achllle