nasa-jpl/open-source-rover #163: PCB Assembly 1.1.9 instruction text mismatch with Table 10

Roger-random, 2019-09-11:
(Related issue: #162 )

In the instruction text for section 1.1.9, parts references do not match their description in Table 10.

1. 6-32 standoffs are not T4. Was T3 intended?
2. 4-40 standoffs are not T11 per Table 10, but T5 is actually correct. (See #162 )
3. 2-56 standoffs are not T6. Unsure if the intent is T8, T11, or another part. (See #162 )
4. M2.5 standoffs are not T7. Was T10 intended?

Achllle, 2020-01-19:
I wish I had seen this before. @mikcox this and the related issue seem to be high priority since it means that people will not have enough parts (and be confused)

Achllle, 2020-01-19:
Instead of making a new issue, I'll add this on here as well:

closed 2023-09-25 by Achllle