nasa-jpl/open-source-rover #167: Body front/back laser cut panels are at different scale

Roger-random, 2019-09-11:
RE: back_plate.DXF and front_plate.DXF submitted https://github.com/nasa-jpl/open-source-rover/commit/831708420ae1a5db06d0873cb76e857595a329ee

These two files appear to have been generated at different scales. When imported into Lightburn laser cutter software, back_plate.DXF is interpreted to be the correct size. However, front_plate.DXF comes in super tiny, approximately 2% of correct size.

Workaround in Lightburn, may be available in other laser cutter software:
* Import back_plate.DXF
* ungroup the paths.
* Delete volt meter rectangle and power switch cutouts. 
* Use resulting cutout-free panel as front plate.

JHPHELAN, 2019-09-22:
Could this be a units problem?  Metric/Imperial, cm/mm?
JHP

Roger-random, 2019-09-23:
Not sure but at the moment I think it is something in Lightburn. When I import these files into Inkscape both files are at the same (correct) scale. Closing now, I'll reopen if I can pin it down to something we can change in rover DXF.

closed 2019-09-23 by Roger-random