nasa-jpl/open-source-rover #172: Rover 6" PVC Neck Too Long For Cable E26

Roger-random, 2019-09-16:
(Related issue: #168) 

Head Assembly Build PDF, section 3.1:

> You can make your rover's neck as long as you'd like, but for reference we cut our "neck" PVC pipe to be roughly 6 inches long.

The neck is actually constrained by the length of cable (E26) connecting main board to Arduino shield, which is 12 inches (~300cm) long.

In the first iteration I cut a 7-inch long PVC pipe. I found the cable could not reach its connector.

[image]

With a 5 inch (~128 cm) length of pipe, the cable could reach, but only barely which places a lot of stress on cable and connectors.

It is far better to cut PVC down to 4.5 inch (~115cm) for some slack and reduce tension.

[image]

Alternate solution: order a longer cable E26. See also related issue #168

aktfire747, 2019-09-17:
It is far better to cut PVC down to 4.5 inch (~115cm) for some slack and reduce tension.

Btw 4.5 inches is 11.43 cm, not 115 cm. The dot makes all the difference. Same with accuracy of conversion of imperial to metric and vice versa. :)

Roger-random, 2019-09-17:
Good catch on my typo! I was debating whether to put down 115mm or 11.5cm and ended up with 115cm which is not a satisfactory compromise between the two options. I'll edit the original text.

mikcox, 2020-03-28:
I've updated our language in the build documents to specify a neck length of 4.5 inches.

Thanks for bringing this to our attention!

commit 05fac900 (2020-03-28): Now specifying length of neck PVC.  Closes #172.
closed 2020-03-28 by mikcox