nasa-jpl/open-source-rover #173: Stale osr-rover-code submodule reference

Roger-random, 2019-09-18:
A newly built rover using information as of mid-August, pulling down everything via ```git clone --recurse-submodules```, was unable to run the rover software due to two issues:

1. To drive LED matrix, ```ros.launch``` kicks off screen.py when it should be arduino_comm.py. Looks like this has been resolved in the osr-rover-code repo but open-source-rover repo still points to now-stale rover-code @5858da7 without the fix.

2. To talk to Roboclaw, roboclaw_wrapper.py uses hard coded device path ```/dev/ttyAMA0``` instead of pulling from ROS parameter which specified ```/dev/serial0```. This is already tracked in issue 34 of the osr-rover-code repo and, once addressed, will need another refresh of submodule reference in open-source-rover repo.

### Workarounds

1. Edit ```ROS/osr_bringup/launch/ros.launch``` line 20 from
```<node name="led_screen" pkg="led_screen" type="screen.py" respawn="true" />```
To
```<node name="led_screen" pkg="led_screen" type="arduino_comm.py" respawn="true" />```

2. Edit ```ROS/osr/scripts/roboclaw_wrapper.py``` line 28 from
```self.rc = Roboclaw("/dev/ttyAMA0",115200)```
To
```self.rc = Roboclaw("/dev/serial0",115200)```

### Symptoms

1. Because ```ROS/led_screen/scripts/screen.py``` initializes ```out_msg[]``` with invalid negative values, and never had a chance to update it with valid data before calling ```send_msg()```, it will fail with the following error at first:

```
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 222, in <module>
    main()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 216, in main
    led.send_msg()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 163, in send_msg
    message.append(val)
ValueError: byte must be in range(0, 256)
```

Once that failure occurs, serial communication between Pi and Arduino is stuck in a bad state and any further attempts will continue to fail repeatedly with this error:

```
Traceback (most recent call last):
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 222, in <module>
    main()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 212, in main
    led = LedScreen()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 49, in __init__
    self.send_init()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 206, in send_init
    if (ord(response[0]) == 67 and ord(response[1]) == 68):
IndexError: string index out of range
```

Once stuck, we'll have to power cycle the Arduino to reset it.

2. Because ```ROS/osr/scripts/roboclaw_wrapper.py``` is talking to the wrong port, when it attempts to communicate:

```Attempting to talk to motor controller128```

It will never receive a response and wait at this point forever.

JHPHELAN, 2019-10-05:
Following issue @RogerRandom:
https://github.com/nasa-jpl/open-source-rover/issues/173

the osr.launch file edited as advised (in all locations):

To drive LED matrix, ros.launch kicks off screen.py when it should be arduino_comm.py. Looks like this has been resolved in the osr-rover-code repo but open-source-rover repo still points to now-stale rover-code @5858da7 without the fix.

Workaround:
Edit ROS/osr_bringup/launch/ros.launch line 20 from
<node name="led_screen" pkg="led_screen" type="screen.py" respawn="true" />
To
<node name="led_screen" pkg="led_screen" type="arduino_comm.py" respawn="true" />

Symptoms
Because ROS/led_screen/scripts/screen.py initializes out_msg[] with invalid negative values, and never had a chance to update it with valid data before calling send_msg(), it will fail with the following error at first:
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 222, in <module>
    main()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 216, in main
    led.send_msg()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 163, in send_msg
    message.append(val)
ValueError: byte must be in range(0, 256)
Once that failure occurs, serial communication between Pi and Arduino is stuck in a bad state and any further attempts will continue to fail repeatedly with this error:

Traceback (most recent call last):
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 222, in <module>
    main()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 212, in main
    led = LedScreen()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 49, in __init__
    self.send_init()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 206, in send_init
    if (ord(response[0]) == 67 and ord(response[1]) == 68):
IndexError: string index out of range
Once stuck, we'll have to power cycle the Arduino to reset it.

When I run:

pi@Fidelity:~ $ roslaunch osr_bringup osr.launch
I get...

... logging to /home/pi/.ros/log/266bdef0-e799-11e9-bbfd-b827eb7e3218/roslaunch-Fidelity-1713.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Fidelity:39217/

SUMMARY
========

PARAMETERS
 * /accel: 0,0,0,0
 * /battery_high: 18
 * /battery_low: 11
 * /baud_rate: 115200
 * /enc_max: 1380,1480,1200,1230
 * /enc_min: 310,345,75,425
 * /joy_node/autorepeat_rate: 1.0
 * /joy_node/coalesce_interval: 0.05
 * /mech_dist: 7.254,10.5,10.5,1...
 * /motor_controller_addresses: 128,129,130,131,132
 * /motor_controller_device: /dev/serial0
 * /qpps: 0,0,0,0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    joy_node (joy/joy_node)
    joystick (osr/joystick.py)
    led_screen (led_screen/screen.py)
    motor_controller (osr/motor_controller.py)
    rover (osr/rover.py)

auto-starting new master
process[master]: started with pid [1723]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 266bdef0-e799-11e9-bbfd-b827eb7e3218
process[rosout-1]: started with pid [1736]
started core service [/rosout]
process[motor_controller-2]: started with pid [1742]
process[joystick-3]: started with pid [1749]
process[rover-4]: started with pid [1754]
process[led_screen-5]: started with pid [1756]
process[joy_node-6]: started with pid [1757]
Attempting to connect to Arduino...
10
[ERROR] [1570298065.444768066]: Couldn't open joystick force feedback!
Traceback (most recent call last):
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 230, in <module>
    main()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 220, in main
    led = LedScreen()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 49, in __init__
    self.send_init()
  File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 212, in send_init
    print str(response[0])          #DEBUG
IndexError: string index out of range
[led_screen-5] process has died [pid 1756, exit code 1, cmd /home/pi/osr_ws/src/led_screen/scripts/screen.py __name:=led_screen __log:=/home/pi/.ros/log/266bdef0-e799-11e9-bbfd-b827eb7e3218/led_screen-5.log].
log file: /home/pi/.ros/log/266bdef0-e799-11e9-bbfd-b827eb7e3218/led_screen-5*.log
[led_screen-5] restarting process
process[led_screen-5]: started with pid [1788]
Attempting to connect to Arduino...
10
repeat...

Power cycling the Arduino fails to resolve the problem.

JHP
USAi Labs (formerly Houston Robotics)

Roger-random, 2019-10-06:
To JHP:

Your error trace

`File "/home/pi/osr_ws/src/led_screen/scripts/screen.py", line 230, in`
`main()`

Indicates ROS is still launching `main()` of `screen.py` and not the correct `main()` in `arduino_comm.py`

Please double check `osr.launch` points to `arduino_comm.py` and not `screen.py`.

JHPHELAN, 2019-10-06:
@Roger-random 
What is the full path to the correct osr.launch?  (_there are at least 2 and I thought I changed both_)
JHP

Roger-random, 2019-10-06:
The correct path will be dependent on your system configuration, fortunately ROS includes an utility to help us find the exact file it would use.

First, make sure you have an editor specified. I personally use `nano` on my Raspberry Pi so

`export EDITOR=nano`

Then, in the _exact same_ environment as you would use to launch OSR code (where you would run `roslaunch osr_bringup osr.launch`) type `rosed osr_bringup osr.launch`. "ROS Edit" will bring up your $EDITOR with the exact same file in package that "ROS Launch" would use.

JHPHELAN, 2019-10-08:
@Roger-random 
Above when you said:
"Workarounds
Edit ROS/osr_bringup/launch/ros.launch line 20 from..."
Did you mean osr.launch instead of ros.launch as osr.launch is what I have.

In EITHER case:
pi@Fidelity:~ $ export EDITOR=nano
pi@Fidelity:~ $ rosed osr_bringup osr.launch
That file does not exist in that package.
pi@Fidelity:~ $ rosed osr_bringup ros.launch
That file does not exist in that package.

When I look for osr.launch:
pi@Fidelity:~ $ find ~/ -name osr.launch
/home/pi/osr/ROS/osr_bringup/launch/osr.launch
/home/pi/osr_ws/src/osr_bringup/launch/osr.launch

pi@Fidelity:~ $ cat /home/pi/osr/ROS/osr_bringup/launch/osr.launch
...
<node name="led_screen" pkg="led_screen" type="screen.py" respawn="false" />
pi@Fidelity:~ $ cat /home/pi/osr_ws/src/osr_bringup/launch/osr.launch
...
<node name="led_screen" pkg="led_screen" type="screen.py" respawn="true" output="screen" />
...even though I KNOW I changed both osr.launch files!
So I changed them AGAIN.
This time roslaunch stops at the usual joystick warning and DOESN'T give me repeat errors on "index out of range".  The LED display has changed from a series of WHITE lines at the top,
to a series of GREEN lines, except the first which goes red through yellow to green.
The display doesn't tell me much, but I'm sure the detailed explanation elsewhere will translate it.]

Any idea why I get the "That file does not exist in that package" error when the files clearly exist and work?

Thanks!
JHP
USAi Labs

ericjunkins, 2019-10-08:
You shouldn't have multiple launch files, only one is necessary, and it should be in whatever workspace is set as your source destination, which you set by editing the ~/.bashrc file

JHPHELAN, 2019-10-08:
my .bashrc reads as per Software Steps:
source /opt/ros/kinetic/setup.bash
source /home/pi/osr_ws/devel/setup.bash
source /home/pi/osr_ws/devel/setup.sh
so the
/home/pi/osr/ROS/osr_bringup/launch/osr.launch
version shouldn't be necessary (rename & test - correct!)
Just from playing I see the 
4th set of 6 LEDs indicate red when a drive motor stalls
5th set of 4 LEDs indicate red when a corner motor moves
will have to explore the code to see what the rest do.
Some brief explanation of the display should be made part of the build instructions.
I'm thinking of finding a way to display simultaneously or alternately a banner advertising our group.
JHP

commit 81da60f7 (2020-03-28): Updated reference to osr-rover-code submodule.  Closes #173.
closed 2020-03-28 by mikcox