nasa-jpl/open-source-rover #175: Corner Steering Angle Calculation Inconsistent With PCB Rev E

Roger-random, 2019-09-20:
RE: Steering angle calculation in file ```Software\rover-code\ROS\osr\scripts\robot.py``` method ```calculateTargetDeg(self, radius)```

This calculation is based on RoboClaw 4 M1 steering the front left wheel, 4M2 steers back left. And RoboClaw 5M1 steeris back right, 5M2 steers front right. This is consistent with Electrical Build PDF Figure 9 (See also issue #174 ) but inconsistent with PCB Rev E which is left-right reversed for the two RoboClaws.

## Symptom:

The calculated Ackermann angles are left/right reversed, causing inside and outside steering wheels to fight each other instead of turning in a coordinated fashion.

## Workaround:

Assign angles according to PCB Rev E locations, which means swapping 1234 to 4321.

Edit lines 126-129 from
```
		ang1 = int(math.degrees(math.atan(self.d1/(abs(r)+self.d3))))
		ang2 = int(math.degrees(math.atan(self.d2/(abs(r)+self.d3))))
		ang3 = int(math.degrees(math.atan(self.d2/(abs(r)-self.d3))))
		ang4 = int(math.degrees(math.atan(self.d1/(abs(r)-self.d3))))
```

To
```
		ang4 = int(math.degrees(math.atan(self.d1/(abs(r)+self.d3))))
		ang3 = int(math.degrees(math.atan(self.d2/(abs(r)+self.d3))))
		ang2 = int(math.degrees(math.atan(self.d2/(abs(r)-self.d3))))
		ang1 = int(math.degrees(math.atan(self.d1/(abs(r)-self.d3))))
```

Alternate workaround: Same mathematical results can be obtained by reversing addition/subtraction of ```self.d3``` instead of changing variable assignment.

ericjunkins, 2019-09-23:
Also dealt with in issue-33 branch of software repo : https://github.com/nasa-jpl/osr-rover-code/commit/29068ba9478443938ec2f8b64a010263c788176b

Roger-random, 2019-09-23:
Great to hear, it reduces this issue to a duplicate of #173

closed 2019-09-23 by Roger-random