nasa-jpl/open-source-rover #176: Wheel Velocity Calculation Left/Right Reversed

Roger-random, 2019-09-20:
RE: Steering angle calculation in file ```Software\rover-code\ROS\osr\scripts\robot.py``` method ```calculateVelocity(self,v,r)```

When turning, the commanded wheel velocities are reversed from what they should be: inside wheels are commanded to turn faster than outside wheels. This may be a consequence of multiple issues upstream (See #174, #175 ) and not necessarily a bug in this math.

## Symptom: 
Depending on traction conditions, one of two things happen during a turn:
1. Rover speed dictated by the slow outer wheels. Inner wheels are scrubbing against the ground, slipping faster than their surface velocity.
OR
2. Rover speed dictated by the faster inner wheels. Outer wheels skid on the surface, turning slower than their surface velocity.

## Workaround:
Reverse the velocity assignment for turning radius greater/less than zero;

Edit lines 105-108 From
```
			if (r < 0):
				velocity = [v1,v2,v3,v4,v5,v6]
			elif (r > 0):
				velocity = [v6,v5,v4,v3,v2,v1]
```

To
```
			if (r > 0):
				velocity = [v1,v2,v3,v4,v5,v6]
			elif (r < 0):
				velocity = [v6,v5,v4,v3,v2,v1]
```

ericjunkins, 2019-09-23:
This was addressed on issue-33 branch of the software repo: https://github.com/nasa-jpl/osr-rover-code/commit/29068ba9478443938ec2f8b64a010263c788176b

Roger-random, 2019-09-23:
Great to hear, it reduces this issue to a duplicate of #173

closed 2019-09-23 by Roger-random