nasa-jpl/open-source-rover #179: Orientation of wheel clamping hub S14A is important and should be clarified and depicted consistently

Roger-random, 2019-10-12:
RE: Wheel Assembly Build Doc PDF

Clamping hub S14A is flat on one side and the other side has a cylindrical protrusion. Diameter of this cylinder is smaller than wheel center, meaning it easily fits inside the wheel which will bolt against the surface of four tapped holes of S14A. As a result, the location of this wheel mating surface would vary by a few millimeters depending on whether S14A is bolted flat-side-in or flat-side-out. Both were depicted in documentation (list below.)

When bolted flat-side-in, with certain Traxxas wheels their mounting screw B5 extends far enough past hub S14A to make contact with face of motor E45 gearbox. In case of light contact, they will acts as brake pads. In case of strong contact, they will prevent the wheel from turning.

If this situation is not detected before motor calibration begins, applying motor power will strip gears in the gearbox.

Recommendation: Make all diagrams consistently flat-side-out and caution rover builders to examine their wheel mounting bolts to ensure their wheels can turn before applying motor power.

In Wheel Assembly Build PDF as of late August 2019:

_Figure 1: Drilling the Wheels_ shows flat side in
_Figure 3: Wheel Step 1_ shows flat side out
_Figure 4: Wheel Step 2_ shows flat side in

darpatel, 2020-01-07:
we also saw that using B4's instead of B5's when bolted flat-side-in solved the issue

ericjunkins, 2020-11-19:
Closing with changing to godiba motors this is not an issue anymore

closed 2020-11-19 by ericjunkins