nasa-jpl/open-source-rover #183: Init Script errors

JHPHELAN, 2019-11-17:


JHPHELAN, 2019-11-20:
Antonio comes through again:

Hi Jim,
The only issue I see, and it makes sense that it does not want to be enabled is the last line of the systemd file,
"Wantedby=multi-user.target"  should be "WantedBy=multi-user.target" 
Basically WantedBy tells Linux if you load the multi-user.target also load this script/service.
Let me know if that helps.
Antonio

Capitalized the B in WantedBy and the error disappeared!
Another typo to fix in the documentation / sample code...

JHP

Achllle, 2020-06-25:
Good catch, I also had to fix this. I'll create a PR for this soon.
Software related issues are a better fit for the [osr-rover-code](https://github.com/nasa-jpl/osr-rover-code) repo by the way. Documentation for the startup service should probably move to there as well.

Achllle, 2020-11-19:
This has been fixed, please reopen on the osr-rover-code repository if the issue persists

closed 2020-11-19 by Achllle