nasa-jpl/open-source-rover #184: Discontinued Part: 1/8" Bore 32 Pitch Shaft Mount Pinion Gear

mikcox, 2019-12-09:
It appears that ServoCity has discontinued the [1/8" Bore 32 Pitch Shaft Mount Pinion Gear](https://www.servocity.com/0-125-1-8-bore-32p-shaft-mount-pinion-gear).  Though this part is available through other resellers, we will need to identify a replacement part.

(on initial search, perhaps [this one](https://www.servocity.com/0-125-1-8-bore-32p-shaft-mount-pinion-gears-steel) would work well)

Achllle, 2020-02-27:
Can confirm that the [ServoCity](https://www.servocity.com/0-125-1-8-bore-32p-shaft-mount-pinion-gears-steel) is a working replacement.

mikcox, 2020-02-27:
Excellent, thanks for the confirmation @Achllle !  I'll get that into the master parts list ASAP.

apollokit, 2020-03-16:
Just to be clear, it's the link above (servo city) with 16 teeth (that choice doesn't get captured in the link).

mikcox, 2020-03-28:
Sorry this slipped off my radar!  I just committed a fix with the updated link and cost, and included a note about the 16 teeth option (and part number).

Thanks @apollokit and @Achllle for finding this!

commit a3564ec0 (2020-03-28): Updated discontinued part.  Closes #184
closed 2020-03-28 by mikcox