nasa-jpl/open-source-rover #185: Need one more part S18 in master parts list

Achllle, 2019-12-15:
The rocker-bogie assembly needs two S18 tubes while the differential pivot assembly needs four. However, the master parts list only specifies to purchase five total.
Happy to make a PR to fix this if I'm not overlooking something

Achllle, 2019-12-15:
Upon closer inspection, it seems like the differential pivot only requires two 4in tubes as opposed to four and part S18 on page 5 really is part S18A from cutting?

mikcox, 2020-03-28:
@Achllle your second observation is correct:

- S16 (the 3-foot aluminum tube) is cut down into 2 parts: a 15" piece (which is then referred to as S16A) and a 13.5" piece (which is then referred to as S16B).
- S18 (the 4-inch aluminum tubes) are cut down to 3-inch lengths.  The 3-inch tube is then referred to as S18A.  2 of these cut S18A tubes are used in the differential pivot, and 2 of them are used in the rocker-bogie.

I will modify one instance where we refer to S16 when we really mean S16A, but otherwise I believe the build docs are consistent.

commit 7b48ef90 (2020-03-28): Updated S16 reference to S16B.  Closes #185.
closed 2020-03-28 by mikcox