nasa-jpl/open-source-rover #187: Control Board Issues

dcschooley, 2020-01-24:
I am making changes to the control board and have run into the following issues:

1. Test point T3 isn't connected to anything. This appears to be a placement error in the 
schematic.
2. There are two J16 connectors (switch and battery out.) This makes KiCad complain about having multiple footprints when you update the PCB from the schematic. If you update the PCB from the schematic, "battery out" gets changed to J15 and the BATT+ connection changes to the unstitched battery. The polarity of the battery out connector gets switched if you tell KiCad to re-associate the footprints by reference, so don't do that.

mikcox, 2020-01-27:
@ericjunkins have you run into this before?  Can you shed any light here?

dcschooley, 2020-01-27:
I've fixed both of these in my local version. I'll do a pull request when I receive the updated boards and can verify that I didn't mess anything up.

There were two copies of J16 originally, and for some reason the PCB layout wanted to update it to J15. J15 is also a duplicate, but there isn't really a problem because it duplicates the battery input connector as a battery output connector. I deleted the extra J15 on the PCB and added a new connector in the schematic. Updating the board then works without KiCad complaining about a duplicate footprint. 

By moving J16 to J15, it looks like the plan was to move the battery output connector from the switched battery voltage to an unswitched (no switch, diode, or resistor) connection. With my version, I kept it unswitched. I added a large trace on the edge of the board to connect J15 and the newly renumbered connector. It would be easy to move this trace to the switch.

My changes allow you to use quadrature encoders on the steering motors and also allow for homing switches. Jumpers are used to switch back and forth from the absolute encoder outputs from the voltage dividers to the quadrature encoders.

dcschooley, 2020-01-27:
I also deleted the extraneous trace and moved C2 as discussed in Issue #177.

ericjunkins, 2020-01-28:
Something must have gotten de-synced when I did an update and the schematic reference designators that caused this. @dcschooley also can you post the schematic for your changes?

dcschooley, 2020-01-28:
Here is the updated schematic. I didn't use nice symbols for J31-J34. That needs to be fixed. Those are the connectors where you use jumpers to select absolute or quadrature encoders. I deleted the test points for signals available on J31-J34. 

[Control Board.sch.zip](https://github.com/nasa-jpl/open-source-rover/files/4123619/Control.Board.sch.zip)

ericjunkins, 2020-01-28:
That schematic is blank when I open it

dcschooley, 2020-01-28:
Ok, trying again. Here is the entire directory with everything but the gerbers. 

[Control_Board_revised.zip](https://github.com/nasa-jpl/open-source-rover/files/4123790/Control_Board_revised.zip)

dcschooley, 2020-01-28:
My plan is to use shorting jumpers on J31-34 on the front of the board to select the encoder type. Unfortunately for the jumpers, the front of the board faces down when mounted in the rover, meaning the jumpers could vibrate loose and fall off of the board. The jumpers probably need to be on the back side. They are under RoboClaw 3 and might not fit that way. Moving them down and to the left above the serial cable connection is a pain. They probably need to be moved to the free space between RoboClaw 4 and RoboClaw 5.

ericjunkins, 2020-01-28:
Oh, no it was my mistake, my computer was opening the .Sch as an altium schematic, not a KiCAD schematic. It opened fine in the KiCAD project. Overall I think it will work fine, I have a few comments:

- Be aware that if you are using the quadrature encoder with your header adjustment that you are driving the output to the op_amp at what you have labeled "RCxENy_DIV". I expect the op-amp can handle that and be fine, but just be aware of it in case.

- Maybe think about doing a set of 3x4 headers instead of 4 individual 3 pin headers, https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/103817-2/A108960-ND/4031292, just easier to install and deal with

- It might be worth adding a noise filtering cap on the limit switch, whenever I use limit switches I always add a small like 0.1uF cap to the signal to help with the mechanical de-bouncing that happens in the switch. I

dcschooley, 2020-01-28:
RCxENy_DIV is the renamed output of the voltage divider circuit. There shouldn't be a connection between RCxENy and RCxENy_DIV unless I messed up the schematic (quite possible) or the builder messes up. RCxENy goes to the RoboClaw and is jumpered to either RCxENy_DIV or RCxENy_RAW. You don't need the voltage divider circuitry without the absolute encoders, so you could also leave out the op amps and the rest of the voltage divider. I like the capacitor and 3x4 header ideas.

dcschooley, 2020-01-29:
This should do it. 

[Control Board F r2.zip](https://github.com/nasa-jpl/open-source-rover/files/4127888/Control.Board.F.r2.zip)

dcschooley, 2020-01-30:
Here is the updated schematic and board layout. Ignore the files I uploaded earlier. 
[Control Board Rev F r2.zip](https://github.com/nasa-jpl/open-source-rover/files/4135699/Control.Board.Rev.F.r2.zip)

apollokit, 2021-04-10:
> I am making changes to the control board and have run into the following issues:
>
> 1. Test point T3 isn't connected to anything. This appears to be a placement error in the
>    schematic.
> 2. There are [...]

Confirming that these items look good in the latest schematic version. The subsequent conversation from this issue seems out of date though, I don't see any net  RCxENy_DIV in the latest version

dcschooley, 2021-04-10:
Are there still traces with that nomenclature? I did a revision of the boards, and actually had them made, where I was using limit switches and the built-in quadrature encoders on the ServoCity motors. This let me have 10 identical motors without having to deal with the absolute encoders. I eventually abandoned that idea after some problems with the limit switches. To make this work where you could use either the standard absolute encoders or the quadrature encoders with limit switches, I used used shorting jumpers to allow you to switch "modes." The input side of the jumpers had the _DIV naming for the outputs of the voltage dividers. Any traces with "_DIV" in the name are probably left over from that era.

apollokit, 2021-04-11:
I said in my comment that I _don't_ see any traces with _DIV. So no, none of that nomenclature seems to be left. I haven't fully groked your changes yet though.

closed 2023-09-25 by Achllle