nasa-jpl/open-source-rover #190: Feedback: 30AWG wire (X3, X4, X5, X6) too difficult to work with.

Roger-random, 2020-03-01:
As of August 2019, the wires for carrying wheel drive motor encoder data were 30AWG wires. For disambiguation there were four colors (X3 through X6). They are intended to connect to side-entry terminal blocks E3 on rover PCB.

Their small diameter are useful for routing through small gaps, helped by the flexible insulator. However, when trying to insert bare wires into E3, their small size is easily bent out of position. Once in position, they are so thin the screw-down terminal can't make solid contact, leading to electrical gremlins and/or wires falling out.

Akin's Australia and US rover build teams both independently came to the same conclusion these wire needed larger and more durable ends. The Australian rover got soldered headers, the USA rover received crimped-on connectors.

__Low-impact solution:__
Switch X3 - X6 to larger diameter wires.

__High-impact solution:__
For a future PCB revision, switch from side-entry terminal blocks to commodity wire-to-board electrical connectors. (JST-XH or similar.) It will require more up-front work to wire up the connector, though note in this case we ended up doing similar work anyway. The payoff is ease of disassembly for diagnosis and repair. For each wheel, one single plug by hand versus 6 individual wires and screwdrivers.

(Or do both, as they are not mutually exclusive.)

------

Log from Australia rover build team:

[image]

------

Photo of USA rover PCB:

[image]

ericjunkins, 2020-07-06:
Looking at making this change to upgrade the wire to be more accomidating to work with. Thinking about the following cables:
@Achllle I think I saw something where you were discussing motor currents and wire current, I tried digging through a few places I thought that discussion was and couldnt find it. Feel free to weigh in. 

| Purpose | AWG | Link |
|:----------| :-----| :-----|
| Motor + | 16    | https://www.digikey.com/product-detail/en/cnc-tech/1569-16-1-0500-004-1-TS/CN645R-25-ND/9450426 |
| Motor  - | 16    | https://www.digikey.com/product-detail/en/cnc-tech/1569-16-1-0500-001-1-TS/CN642B-25-ND/9450423 |
| Encoder + | 24 | https://www.digikey.com/product-detail/en/nte-electronics-inc/WH24-06-25/2368-WH24-06-25-ND/11646871|
| Encoder -  | 24 | https://www.digikey.com/product-detail/en/nte-electronics-inc/WH24-05-25/2368-WH24-05-25-ND/11646835 |
| Encoder A | 24 | https://www.digikey.com/product-detail/en/nte-electronics-inc/WH24-04-25/2368-WH24-04-25-ND/11647079 |
| Encoder B | 24 | https://www.digikey.com/product-detail/en/nte-electronics-inc/WH24-09-25/2368-WH24-09-25-ND/11647824 |

dcschooley, 2020-07-07:
What's driving the change to 16AWG wire for the motors? It seems like overkill. 4ft of 20AWG has a voltage drop of 0.25 volts or so at 4A. 16AWG is less than 0.1 volt. The PWM going into the motors will change things a bit, but I'm not sure how much. Going with wire that size limits connector options.

Achllle, 2020-07-08:
It's not the voltage drop that's the issue, it's the current draw from the motors. The idea is to replace the default motors with beefier ones (which a lot of folks have done already) and current draw for a single motors could peak ~10A or more which is dangerous. Perhaps 18AWG is a middle ground?

dcschooley, 2020-07-08:
Voltage drop and heating due to current are two sides of the same coin. When sizing the wires, steady-state current matters more than transients. Except for the protective wire braid, our motor wires are basically exposed to air, as opposed to being enclosed in a wall or in conduit. That helps with any extra heat from transients and we are not worried so much about the total heat produced in the length of wire. The default battery is good for 8A at 16.8 V, so less than 135W total into the rover for a fully charged battery. 120W is probably a good number for when you've been driving for a while. That gives you 20W average for each drive motor. (The multimeter thing is good for 20A, so we can ignore that as a limit.) 

The worst case is if the motors can pull the 20W at 6V, so that gives you a bit over 3A. My estimate of worst-case steady-state current is 4A. That's 4A into each motor as you drive across the prairie. At that current, assuming I did my math correctly, the power loss in each motor wire is 1W for 20AWG, and some fraction of that for bigger wires. At 10A, it's only 2.5W, which will only occur for a short time. I'm using the beefy 45 RPM Servo City planetary gear motors, and 10A into those motors for very long when stalled will destroy them. These motors come with 24AWG wire and JST PH series connectors, and anything much bigger is probably a waste of copper. (The battery has 16AWG wire.) The PH connectors are good for 2A steady state. My first thought when I was looking at the project was that the 20AWG motor wires were overkill, but they do help reduce the losses between the motor controllers and the motors. 

I'll see if I can do a better characterization of the motors from the specs, just to see how they might look.

dcschooley, 2020-07-08:
So here's a screenshot of a table calculating voltage drop and wire losses for 1m length of wire. I got the resistances from some random place on the internet. I think the sweet spot is 20AWG for the motors. 22AWG for the motors would make it easier to find board connectors that would work for both the encoders and motors, but 20AWG will work.

[image]

Achllle, 2020-07-08:
I don't think everyone uses the default battery. I switched to a 9Ah 100C LiPo... The battery leads are 12AWG...

> 120W is probably a good number for when you've been driving for a while. That gives you 20W average for each drive motor.

Aside from using the battery upper limit as the starting point, I don't think I trust dividing that number by the number of drive motors. Loads are often very unbalanced, and it's also common to turn and drive at the same time.

> These motors come with 24AWG wire

Wow that seems crazy, are you sure of that? The [website](https://www.servocity.com/45-rpm-hd-premium-planetary-gear-motor) seems to show 16AWG. I have the GoBilda planetary gear motors and they come with 16AWG wires.

dcschooley, 2020-07-08:
> I don't think everyone uses the default battery. I switched to a 9Ah 100C LiPo... The battery leads are 12AWG...
> 

The leads on your battery are so big because your battery has an extremely high discharge capacity. It appears to be a multi-rotor battery. The same size battery with a lower discharge capacity could get by with smaller wires.

> Aside from using the battery upper limit as the starting point, I don't think I trust dividing that number by the number of drive motors. Loads are often very unbalanced, and it's also common to turn and drive at the same time.

Dividing by six gives you an upper estimate for sizing the drive motor wires. Turning at the same time reduces the amount of current available for the drive motors, but that's a short-term impact and it wouldn't impact the size of the wires. If anything, it would make them smaller.

> 
> > These motors come with 24AWG wire
> 
> Wow that seems crazy, are you sure of that? The [website](https://www.servocity.com/45-rpm-hd-premium-planetary-gear-motor) seems to show 16AWG. I have the GoBilda planetary gear motors and they come with 16AWG wires.

Yes, they have 24AWG wires. The JST PH connectors they come with are limited to 24AWG.

Achllle, 2020-07-09:
> Dividing by six gives you an upper estimate for sizing the drive motor wires.

If you assume that all drive motors draw the same current at all times. When there is more friction on one wheel or it's stuck and the other wheels are spinning freely, that one motor is going to draw quite a bit more.

In my case (and I think with quite a few people who decided to upgrade the battery), since the discharge capacity is so high, there's practically no limit on the 'current available'. For the case of a limited discharge, I agree with your point, but we want to support different batteries and discharge currents.

Regarding the wires on those motors: on the product page in the recommendations section they link the [16AWG wires](https://www.servocity.com/3-5mm-bullet-lead-fh-mc-300mm-length) and on that page they show the original motors so I'm still not convinced 24AWG is a good idea 🙂

dcschooley, 2020-07-09:
You are correct that some motors may be pulling more power than others for whatever reason, but 20W is still a good baseline for steady-state (average) power calculations. So the question is what happens if you start to deviate from normal. 

Here are estimates of torque-speed and power-speed curves for those motors. I just took values from the provide torque-speed curves and calculated the output power. You are not going to get much more than 20W out of these motors no matter what you do. The current requirements are going to go up, but it will require more effort to calculate that and I'm not sure I could do it correctly right now. (I'm not entirely sure I got this right.) You certainly don't want to push more than 7A steady-state out of the RoboClaws for any extended period of time. Transients can certainly be higher than 20W, but as I said before, you really don't care about those WRT the wires from the RoboClaws to the motors. 

If you look on the ServoCity page with the encoder versions of those motors, you will see the 24AWG wires and connectors. These are what you want for the drive motors. 

I'm not saying to use 24AWG, I'm saying to stick with 20AWG unless there is a good engineering reason to make the wires bigger, and I'm not sure there is one. Tradeoffs include getting the wires from the motor to the body and connectors on the board. The blue connectors currently used for the battery connections are only good for 15A steady state. I am updating the PCB and will incorporate better connectors for the motors and encoders. The 24AWG-20AWG range of wire sizes is a sweet spot for connectors. We could use JST XA connectors and have a single connector for each motor/encoder combination. Anything larger than 20AWG means going up to JST VH connectors for the motors and hope they fit while using a different connector for the encoders. I'm also looking at incorporating an PI-interfaceable device for current and voltage measurements (per your request) and the AdaFruit device is only good for 15A steady state. 

[image]

Achllle, 2020-07-09:
Thanks for the thorough analysis. I'm on the same page now!

apollokit, 2021-04-20:
Alright, so is the conclusion here to use 20 AWG for Motor +/-, and 24 AWG for the encoder wires?

closed 2021-06-29 by dcschooley