nasa-jpl/open-source-rover #191: Move software installation steps, software controls to osr-rover-code

Achllle, 2020-04-15:
Since these things are subject to frequent change under development, it might make sense to move anything software (perhaps other than how to install ROS and setup the RPi and Arduino and so on) to the `osr-rover-code` repo. That would avoid having to create and synchronize PRs in two places whenever something is changed. We would simply refer to the other repo when users are done setting up the RPi and Arduino.

The software controls document in this repo is a good example of that. We've recently overhauled the code to have fairly different controls and so two sources of documentation is confusing and wrong. I can create a similar visual explanation for how the calculations for drive and corner motor positions and velocities work now in the osr-rover-code repo.

mikcox, 2020-04-24:
Hey @Achllle!  Thanks for your thought on this.  We believed early on that having all the documentation together in one place was a really important part of the project to make sure it was ingest-able for folks.  To that end, we actually even started our documentation in one huge .pdf file.  After it got to 100+ pages, we realized we had to start breaking it down.

I agree with the sentiment that the software installation steps are usually highly related to the `osr-rover-code` repository.  I'm also trying to balance my desire to have all the documentation in a similarly-structured, easy-to-find place.

I think we can find a happy medium where we still have a front-to-back, "step 1 to step n" flow in a single place (in the `open-source-rover` repo) while we can still refer to .pdf documents in other repos.

Achllle, 2020-04-27:
Just to make sure we're on the same page, did you have a chance to look at the PDF in #192? Unlike I mention above, I just moved some specific steps to `osr-rover-code`, how to set up the Pi and Arduino in general are still left in this repo.

Unlike the mechanical part of things, I think that the software on the rover is still under heavy development, so I think it would help to move things that are still very prone to change to the new repo. Whatever we do, I think enabling speed of development and quality of documentation is paramount.

Achllle, 2020-06-11:
closing because of #192

closed 2020-06-11 by Achllle