nasa-jpl/open-source-rover #193: PCB fragile traces

JHPHELAN, 2020-04-25:
Having recently upgraded the 4 steering motors to the new specs and all 6 drive motors to the REV Robotics HEX motors I re-tuned them all using Basicmicro Motion Studio and they checked out ok (after I fixed a couple Molex connections that came loose.)  But when I tested it with the Xbox controller, the left side steering was paralyzed.  Logic demanded that it was a problem between the Pi and the RoboClaws.  Continuity check between the Pi and the PCB was intact.  Since communication with the RCs through the PCB was serial, there had to be a problem near RC5/132 that controlled steering motors 3 and 4.  [BTW, Section 4.2 figure 9 seems mis-labeled to me.  Front/Back are reversed and the steering motors are switched right/left when viewed from above.]  Examining the PCB I found a blown trace near RC5.  Above RC5 and next to RC3 in attached pic.  The two red wires near the fried trace are jumpers from a previous blown trace to the LED display.  The second pic shows the repair (prior to cleaning up the flux and solder stains).  Looking at the Gerber of the PCB (I have Rev D.  Not sure what the changes are to the current Rev E) it seems that the traces are very narrow and wouldn't be able to handle much current making them more susceptible to failure.  Would suggest increasing the width of all traces as much as possible on any future revision.  (along with a number of other changes, but that's a different story.)
[image]
[image]

ericjunkins, 2020-04-26:
Can you link the motors you use please. If you are using a motor that pulls significantly more power than the initial ones then yes I could see that happening.

JHPHELAN, 2020-04-26:
The blown trace is not a motor power trace, but the serial path to the RC.
The motor is the standard OSR upgraded steering motor: 
[https://www.servocity.com/45-rpm-hd-premium-planetary-gear-motor](url)

ericjunkins, 2020-04-26:
The traces you show jumped in the above picture do not go to the roboclaw, nor are they serial lines. And i don't think what is jumped herewith wires is correct as well. It appears that the section you are jumping is for the J22 Connector, J10 connector, and C7 capacitor I believe. The First picture shows  the follwing jumpers:

ericjunkins, 2020-04-26:
The traces you show jumped in the above picture do not go to the roboclaw, nor are they serial lines. It appears that the section you are jumping is for the J22 Connector, J10 connector, and C7 capacitor I believe.

 J22 pin 4 (+5V) to

JHPHELAN, 2020-05-06:
@ejunkins 
Sorry for the slow reply.  The blown trace is to J22 the +5V which powers the encoder for the Left front steering motor (mis-labeled Right in this version D of the board) so the RoboClaw couldn't read the encoder.

[image]

ericjunkins, 2020-05-06:
@JHPHELAN right that's the trace I mentioned above. I am wondering if there was a short in your system, especially if you look closely at this picture, you can see that you have a bit of exposed wiring on the +V line, which is directly next to the GND
[image]

In order to blow the trace that did you would have to pull in excess of 1.5Amps, which that encoder would never come anywhere close to without a short somewhere.

ericjunkins, 2020-06-25:
Closing from non-issue

closed 2020-06-25 by ericjunkins