nasa-jpl/open-source-rover #196: DROK current+voltage sensor replacement

Achllle, 2020-07-02:
We currently use the DROK multimeter with screen. There's no interface other than the display for it so we can't read the values from the Raspberry Pi or Arduino.
It would be useful to access these values so we can log them. Brief current spikes might not show up and are hard to see. We could still display the real-time values on the LCD display.
You can get a digital sensor for under $10.

dcschooley, 2020-07-09:
It will be easy to add this: <https://www.adafruit.com/product/4226>. I'm assuming this part is decent. Is there anything better?

This device would become part of the incoming power circuit, so it would be a requirement. I'm wondering if it would make sense to replace the DROK with a small display driven by this thing and the PI.

ericjunkins, 2020-07-09:
@dcschooley Right I was thinking about that too, that i dont necessarily want to get rid of entirely a way to display voltage and current etc to the user, would be good to have both the ability to read  the values like with the adafruit part you mentioned, and have a small display

dcschooley, 2020-07-09:
I think you definitely need to have a user-viewable display. Part of it is educational, the other is you simply need to know what is happening with the rover. Having the DROK is useful during testing so that you don't let the battery voltage to get too low. Losing that would be slightly annoying if the PI wasn't up and running yet. One advantage of a controllable display is that the PI could display warning messages if the battery voltage gets too low or the current gets too high, preferably in a different color.

dcschooley, 2020-07-09:
Here's a possibility for socketing the power connection for the Adafruit device: <https://www.digikey.com/product-detail/en/samtec-inc/HPF-02-01-T-S/SAM9491-ND/6678228> and <https://www.digikey.com/product-detail/en/samtec-inc/TSW-202-07-T-S/SAM10858-ND/2686027>

apollokit, 2021-04-09:
> I think you definitely need to have a user-viewable display. Part of it is educational, the other is you simply need to know what is happening with the rover. Having the DROK is useful during [...]

Another potential option here is that we could just make a little web app that is easy for people to run, which collects and displays telemetry from the rpi. Definitely not as reliable as a built-in display, and wouldn't currently work when out of wi-fi range.

apollokit, 2021-04-09:
Why don't we just include the [TI chip](https://www.ti.com/lit/ds/symlink/ina260.pdf) on the control board directly for rev F? Seems like a more straightforward solution rather than having a daughter board that can mechanically fail.

dcschooley, 2021-04-09:
> Why don't we just include the [TI chip](https://www.ti.com/lit/ds/symlink/ina260.pdf) on the control board directly for rev F? Seems like a more straightforward solution rather than having a daughter board that can mechanically fail.
> 
> Plus I think it would be fun to do the work of including it :)

That's a really good idea. Unfortunately you can only get it in a surface mount package. Doing surface mount would be a bit much to ask newb solderers.

ericjunkins, 2021-04-09:
Yeah I guess it depends on where we sit on the line of difficulty vs functionalities. It's a Tssop 16 package which is at least reasonable, something like qfp64 would be out of the question.

https://jeelabs.org/book/1513c/index.html

Here's a little breakout board I've used things similar to this for breadboard testing of surface mount components. One could potentially include these, and have it attach via the 0.1 inch headers, that way you have essentially unlimited tries and aren't ruining the main pcb

dcschooley, 2021-04-09:
But in that case wouldn't it make the most sense to just use the AdaFruit board?

dcschooley, 2021-04-09:
Maybe we just go ahead and solder the Adafruit thing into the PCB like nature intended. Yes, you do have to unsolder it if it goes bad, but it's just a few pins.

apollokit, 2021-04-10:
ah yes, agreed on the surface mount aspect. I didn't think about that.

@dcschooley what do you mean by "solder the adafruit thing"? solder the whole breakout board?

dcschooley, 2021-04-10:
The Adafruit board comes with two pin headers that you solder onto the board in the orientation you want. One is low current signaling that can go into any appropriate socket and the other is for the high current connection. There was some discussion early on about this part regarding how to connect it to the PCB. The easiest way would be to solder the high-current pins directly to the main PCB. The ina260 would be permanently attached at this point. The downside is what happens if the ina260 craps out and you have to unsolder it. I did some checking on the Dici-Key web site and found some socket/pin combinations that were supposed to work together and would also support the necessary current. The ones I ordered did not fit together correctly, even though Digi-Key said they were supposed to. I'm sort of at the point of saying solder the whole thing down and be done with it. Fixing a blown chip would still be easier than trying to deal with the surface mount.

JHPHELAN, 2021-04-11:
I like the DROK and the Adafruit breakout board.  I rearranged my power sequence to:
battery/wall wort > lighted push button > DROK > remote kill switch > PCB.
This way I can see if the battery is adequate before turning on the system with the kill switch.
Surface mount tech is a bit much to ask of us amateurs.  
I like the ease of the Adafruit breakout board so we should make a place for it on the PCB.  
The next question is how to handle the 8 signal pins?  Jumpers/cable to pins on the secondary ribbon cable connector?
Route them through the PCB to appropriate pins on the Pi?  Which pins?  We should standardize this to avoid confusion with future as yet unknown additions.  What software?  Is there a ROS node for this?  What kind of display?  The current LED display is crude and power hungry.  Use the HDMI port?  The Adafruit 2407 7" HDMI works well.  [https://www.adafruit.com/product/2407](url) or one of their several LCD displays such as [https://www.adafruit.com/product/2050](url)?  A high-res display along with a camera would make for real telepresence.  There are apps to make a cell phone into a USB display or camera.  Don't know if they'll both work at the same time.

Achllle, 2023-09-25:
We now have the DROK and the INA260 so I consider this resolved.

closed 2023-09-25 by Achllle