nasa-jpl/open-source-rover #205: Add filter capacitors to quadrature encoder connections

dcschooley, 2020-07-27:
The Servo City HD Premium Gear Motors with encoders have a capacitor acting as a snubber on the encoder board. This capacitor couples the 20kHz PWM voltage from the RoboClaws into the encoder signals. The 20kHz noise causes errors, showing up as extra pulses, when the RoboClaws read the encoders. BasicMicro recommends 0.01uF capacitors on the encoder lines to filter out the 20kHz noise. Locations for these capacitors need to be added to the control PCB. 

The snubber circuit is a poor design that supposedly shows up in similar motors that have built-in encoders. The extra capacitors can be considered optional for motors that do not need them, but adding the filtering should not impact motors that do not need them, and the extra filtering is not a bad idea.

The 0.1uF capacitors already in use on the board are too large for this purpose and filter out the encoder signals entirely at high motor speeds. The 0.01uF capacitors eventually specified for the project should have a different form factor to keep them from getting mixed up with the 0.1uF capacitors.

SallyTYT, 2022-09-28:
> The Servo City HD Premium Gear Motors with encoders have a capacitor acting as a snubber on the encoder board. This capacitor couples the 20kHz PWM voltage from the RoboClaws into the encoder [...]

0.01uF capacitors on the encoder lines really help reduce the jitters in speed measurements on the RoboClaw. Before I added the capacitors, there were huge jitters in the speed measurements on my 7# and 8# drive motors, leading to the velocity control not working. And the capacitors helped solve this issue.

Achllle, 2023-09-25:
I take it that this issue is solved with the latest PCB design for v2. If not, let me know and we can reopen.

closed 2023-09-25 by Achllle