nasa-jpl/open-source-rover #210: Software Steps / Setting up ROS environment: $ROS_DISTRO;  rosdep install out of order

JHPHELAN, 2020-09-28:
[Note: "tilde" replaced by "home" to prevent weird formatting of text]

4 Setting up ROS environment and building the rover code
4.1 Setup ROS build environment
First we'll create a workspace for the rover code.
Create a catkin workspace directory, which will contain all ROS compilation and 
source code files, and move into it
	ubuntu@ubuntu:home$ mkdir -p home/osr_ws/src && cd home/osr_ws

Source your newly created ROS environment
	ubuntu@ubuntu:home/osr_ws$ source /opt/ros/**$ROS_DISTRO**/setup.bash
	-bash: /opt/ros//setup.bash: **No such file or directory**

**$ROS_DISTRO needs to be replace by “melodic”**
	ubuntu@ubuntu:home/osr_ws$ source /opt/ros/**melodic**/setup.bash
	[no error this time]

4.2 Clone and build the rover code
For this section, you'll be working with the version control software git. Now's a good time to read up on how that works if you're new to it and make a GitHub account! In the newly created catkin workspace you just made, clone this repo:
	ubuntu@ubuntu:home/osr_ws$ sudo apt-get install git
	ubuntu@ubuntu:home/osr_ws$ cd home/osr_ws/src
	ubuntu@ubuntu:home/osr_ws/src$ git clone https://github.com/nasa-jpl/osr-rover-code.git
install the dependencies
	ubuntu@ubuntu:home/osr_ws/src$ rosdep install --from-paths src --ignore-src
	**given path 'src' does not exist**

**Previously cd .. came first, then rosdep.  Let’s try that:**

install the dependencies
	home/osr_ws/src$ cd ..		**BACK UP one dir to get to home/osr_ws dir to do rosdep**
	home/osr_ws$ rosdep install --from-paths src --ignore-src
	home/osr_ws$ catkin_make
	ubuntu@ubuntu:home/osr_ws/src$ cd ..
	ubuntu@ubuntu:home/osr_ws$ rosdep install --from-paths src --ignore-src
	All required rosdeps installed successfully
	ubuntu@ubuntu:home/osr_ws$ catkin_make
	...
	[100%] Built target osr_msgs_generate_messages
add the generated files to the path so ROS can find them
	ubuntu@ubuntu:home/osr_ws$ source devel/setup.bash

Achllle, 2020-10-03:
@JHPHELAN I don't understand what you are trying to convey here. Please edit the issue and describe more clearly what is wrong.

Achllle, 2020-10-03:
I formatted it for you. Also please note that this is also an issue for the other repository. I updated the formatting for you so please take a look at what that looks like. Closing here, please reopen on the other repo with proper formatting.

Issues are a way to convey that something is wrong or could use improvement. In order to make them actionable, they need to be clear and concise, so avoid copying instructions, and instead linking them.

closed 2020-10-03 by Achllle