nasa-jpl/open-source-rover #212: Port Software Controls doc to markdown

apollokit, 2020-10-14:
The [Software Controls](https://github.com/nasa-jpl/open-source-rover/blob/master/Software/Software%20Controls.pdf) doc should be ported to markdown and moved to the [rover code](https://github.com/nasa-jpl/osr-rover-code) repo.

SConaway, 2020-10-14:
Making a sample comment per https://github.com/nasa-jpl/open-source-rover/issues/202#issuecomment-708590686.

I can do this over / by the weekend if you want, just let me know.

apollokit, 2020-10-15:
Sweet, got you assigned. 

No rush on this. Sometime in the next few weeks, it's not pressing. I'm thinking something like a 2 or 3-month time span for all the docs.

Achllle, 2020-10-15:
TBH I might be the one to do this, @apollokit . The current instructions do not reflect the actual controls.

SConaway, 2020-10-15:
Good idea. I don't want to waste my time and then have it all get redone

apollokit, 2020-10-15:
Sounds good, reassigned.

Achllle, 2021-02-14:
I made a [separate issue](https://github.com/nasa-jpl/osr-rover-code/issues/127) in the code repo to add the documentation from here. Once that's done, this can be deleted. For now it's not in anyone's way and it's clearly stated that the document is outdated.

closed 2021-02-14 by Achllle