nasa-jpl/open-source-rover #225: Gobilda adapter pieces missing from build instructions

JHPHELAN, 2020-11-07:
In the Wheel Assembly instructions:

https://github.com/nasa-jpl/open-source-rover/tree/master/Mechanical/Wheel%20Assembly

Assembling the base wheel mounts / Prepare the motor channel:

The threaded plate S46 and pattern adaptor S48 are not listed in the parts list on that page.

Achllle, 2020-11-15:
With the change to markdown, it should be much easier to help fix this! Happy to take this one, but if you'd like to learn how to contribute, I can guide you through the steps to do it. Let me know!

mattbhenley, 2020-11-17:
@Achllle I would love to take this one - trying to get into contributing more often. Where would I be able to find these images for the parts? I would then throw them in the readme and create a pull request. Let me know if that sounds correct! Thanks for the great documentation

Achllle, 2020-11-19:
Awesome, I assigned you to this one - thanks for stepping in! We're using Markdown for that document (other documents are still in LatTeX, we're slowly transitioning away from that) so you can see how it references current images.

Images of parts are in [root/images](https://github.com/nasa-jpl/open-source-rover/tree/master/images), images related to instructions are in each folder/images.

Let me know if you have questions!

mattbhenley, 2020-11-19:
@Achllle https://github.com/nasa-jpl/open-source-rover/pull/228 

Here's my PR. Hope that does the trick.

Achllle, 2020-12-09:
closed with the above PR

closed 2020-12-09 by Achllle