nasa-jpl/open-source-rover #238: Is the link to Master Parts List broken?

elidica, 2020-12-29:
I gt a 404 message.  Thanks!  Would love to try this with my 9th-grade students.

ericjunkins, 2020-12-30:
What did you get a 404 on? A particular part on the parts list? Happy to help

SConaway, 2020-12-30:
@ericjunkins not @elidica, but I think they are referring to the parts list link in the README. The link is pointing to a non-existent file. See that attached screenshot for where: 
[image]

elidica, 2020-12-30:
Yes, exactly.  The link to the excel document pops up on hover but I get a
404 error message when I click on it.  Thank you!

-- 
Elizabeth Nye Di Cataldo
*she/her/lei*

-- 
-----------------------------------------------------------------
Ai sensi 
del D.Lgs. 196/2003,  del Regolamento UE 2016/679 in materia di protezione 
dei dati personali, e dell'l’articolo 616 del codice penale, si precisa che 
le informazioni contenute in questo messaggio sono riservate ed a uso 
esclusivo del destinatario. Qualora il messaggio in parola Le fosse 
pervenuto per errore, La invitiamo ad eliminarlo senza copiarlo e a non 
inoltrarlo a terzi, dandocene gentilmente comunicazione al 
ststephens@sssrome.it <mailto:ststephens@sssrome.it> o telefonando al 
+39065750605. Grazie.
-----------------------------------------------------------------
In 
accordance to the Legislative Decree No. 196/2003, GDPR UE 2016/679, and 
article 616 Penal Code, you are hereby informed that this message contains 
confidential information intended only for the use of the addressee. If you 
are not the addressee and/or have received this message by mistake, please 
delete it and notify us immediately at ststephens@sssrome.it 
<mailto:ststephens@sssrome.it> or +39 06 5750605. Any dissemination, 
distribution or duplication of this communication without prior written 
consent from St. Stephen's School is strictly prohibited. Thank you.

SConaway, 2020-12-30:
@ericjunkins commit 319d472899e02d0f40c597268abe76a9c2257dd5 removed `master_parts_list.xlsx` that the readme is referring to

ericjunkins, 2020-12-30:
@SConaway thanks for digging up where that happened. @Achllle it looks like you made commit 319d472899e02d0f40c597268abe76a9c2257dd5 which removed the masters_parts_list, I didn't see when this happened. What you removed wasn't a duplicate of the master_parts_list_raw.csv, which is under the parts list folder. The raw was used by the python script to compare quantities and have override values for certain things, ie. if you ended up needing 95 screws and it came in a pack of 100, override that to buy 2 packs of 100 etc. 

From our recent discussion about moving to the README format from the latex files I believe we decided not to put effort into fixing and maintaining that python script, so we should probably decide what we would like to do with the parts list now then, and fix the link from the main readme

ericjunkins, 2020-12-30:
@elidica You can take a look at the parts list at https://github.com/nasa-jpl/open-source-rover/blob/master/parts_list/master_parts_list_raw.csv to get parts :)

elidica, 2020-12-30:
@ericjunkins @SConaway  Thanks so much!  I am in an international secondary school in Italy and am hoping to use much of our Tetrix kits to work from to start with.  Appreciate the support!

Achllle, 2021-01-02:
> The raw was used by the python script to compare quantities and have override values for certain things, ie. if you ended up needing 95 screws and it came in a pack of 100, override that to buy 2 packs of 100 etc.

So all the information is in the raw file, correct? If so, the fix in #238 should do the trick.

PR #239 (2021-04-07): Fix broken references to xlsx files + update instructions
Close #238
closed 2021-04-07 by Achllle