nasa-jpl/open-source-rover #250: 9 x 12 top and bottom ServoCity plates discontinued

Achllle, 2021-02-25:
@dcschooley found out they are discountinued.

One idea is to replace with two 9 x 6 plates and connectors.

Achllle, 2021-02-25:
* YouDoItElectronics [still has it in stock](https://www.youdoitelectronics.com/actobotics-9-x-12-aluminum-pattern-plate-585006)
* British website [still has it in stock](https://www.active-robots.com/9-x-12-aluminum-pattern-plate.html) (for now)

These will likely be out of stock as well soon

dcschooley, 2021-02-25:
Using two 4.5x12 inch plates is doable. The hole for the rocker-bogie pivot will be on the seam, which means you are grinding the hole instead of drilling it. Materials for cutting the top and bottom pieces out of acrylic would cost $20-$35, depending on color and where you buy it. (You would need two 12x20 sheets.) With another sheet you can do the entire body in acrylic. I don't know what it would cost to have someone like Sculpteo make them. 

[image]

closed 2021-07-02 by dcschooley