nasa-jpl/open-source-rover #264: fix formatting and numbering in electrical build doc

apollokit, 2021-04-20:
It doesn't look good right now. fix it.

https://github.com/nasa-jpl/open-source-rover/blob/master/electrical/electrical_build.md

gyanprakash0221, 2021-04-21:
I want to work on this issue. Can you guide me through what needs to be fixed @apollokit

apollokit, 2021-04-21:
@gyanprakash0221 great! 

I really just want more consistent sizing of all the images, similar to the [pcb assembly readme](https://github.com/nasa-jpl/open-source-rover/tree/master/electrical/pcb_assembly)

some specifics:
- inconsistent image sizes in table 3
- inconsistent image sizes in table 4
- inconsistent image sizes in table 6
- some images are too big, for example figure 5

Looks like the file is set up pretty well already. You should just be able to play with the image sizes in the markdown, for example `<img src="../images/components/electronics/E26.png" width="25%">`

You can make the changes and open a PR, and we can iterate on it there.

gyanprakash0221, 2021-04-24:
sure sir, I will be doing all the changes required in the process to make the doc more readable

Achllle, 2023-09-25:
closing since no longer relevant with latest update

closed 2023-09-25 by Achllle