nasa-jpl/open-source-rover #269: buy rover's materials

alsalamcse, 2021-05-03:
hi 
i am leading intersting project in my school  for builing this rover.

please tell me how can i buy the full stuff of the rover matrials? 
trhnaks

please contact me by email: samih1079@gmail.com

JHPHELAN, 2021-05-03:
@alsalamcse 
There is no "kit" with all the parts.  
Part of the learning experience of project management is sourcing parts.  Depending on where you live your sources might have to be different.
On the first page is the Parts List [https://github.com/nasa-jpl/open-source-rover/blob/master/parts_list/master_parts_list_raw.csv](url)
Read it from the screen and scroll to the right to find the quantities and price.  The vendor web site is listed.  Better to download the .csv file and import to your local spreadsheet and sort by vendor.  Once I ordered the part, I changed the font color to orange.  Once arrived to green.  If out of stock or other problem, to red.  
I like to organize parts in various size zip-lock bags.  You can see pix of this and our  building steps and techniques in the In-Progress Builds gallery for Houston Robotics (now USAi Labs) on the Tapatalk Forum.  Others have many more.  A GREAT Youtube tutorial for building is Leo's: [https://www.youtube.com/watch?v=DxgcTCMTy-c&list=PL6Tmb7sYBH6feflfsXgHLtBspMLC-_gxl](url)].  The kid is AMAZING!  
When we first started, the projected cost was about $2500.  Not sure what it is now.  But with upgrades, replacements, changes, custom enhancements, tools, accessories, etc. I'm sure we spent $3000.  As part of the first group of builders (we started the week it was announced) with problems,  changes, part delays and life getting in the way, it took us exactly a year til our rover took its "first steps" here: [https://www.youtube.com/watch?v=2Wdgss0q63s](url). 
Be sure to join the TapaTalk forum [https://www.tapatalk.com/groups/jpl_opensource_rover/](url)
and the Slack forum [https://app.slack.com/client/T01CMHZLGPL/C01CTNRDERJ](url)
and read the most recent posts to avoid problems.  For immediate help, use the Slack forum.
What part of the world are you in?
Enjoy!
Jim Phelan

Achllle, 2021-05-06:
Please use either of the forums for such questions. The issue tracker is for problems with the code/files or feature requests.

closed 2021-05-06 by Achllle