nasa-jpl/open-source-rover #271: Update submodule paths

sethfischer, 2021-05-05:
# Problem

The following commands fail to update the Git submodules:

```
git submodule init
git submodule update
```

After cloning the project with `git clone https://github.com/nasa-jpl/open-source-rover.git` we should have the directories that contain the submodules (`software/android-app`, and `software/rover-code`), but none of the files within.

Then `git submodule init` should initialize the local configuration file, and `git submodule update` should fetch the data from [nasa-jpl/osr-android-app](https://github.com/nasa-jpl/osr-android-app) and [nasa-jpl/osr-rover-code](https://github.com/nasa-jpl/osr-rover-code).

[More information on Git submodules](https://git-scm.com/book/en/v2/Git-Tools-Submodules).

# Solution

1. Update paths in `.gitmodules`.
2. Commit the submodules.

sethfischer, 2021-05-06:
If the hardware and software repos are tightly coupled then Git submodules are a great way to go. However, at this stage of development this appears not to be the case.

As the submodule configuration is currently incorrect I think it would be safe to assume that the majority of people are not using submodules and prefer to work without them. If that is the case, I can:

1. Update #272 to remove both the `nasa-jpl/osr-rover-code` and `nasa-jpl/osr-android-app` Git submodules.
2. For those who are using Git submodules I will provide instructions on how to remove them from their local repo (as a comment on this issue).

Achllle, 2021-05-07:
Sounds like a plan! Thanks

sethfischer, 2021-05-07:
I have closed #272 and opened #277 .

If installed follow this process to remove local Git submodules:

1. Merge in nasa-jpl:master to which removes the `.gitmodules` file.
2. Delete the following sections from `.git/config`:

```
[submodule "Software/android_app"]
        url = https://github.com/nasa-jpl/osr-android-app
        active = true
[submodule "Software/rover-code"]
        url = https://github.com/nasa-jpl/osr-rover-code
        active = true
```

3. Run `git rm --cached Software/android_app` and `git rm --cached Software/rover-code`.
4. Delete the now untracked submodule files with `rm -r Software/android_app` and `rm -r Software/rover-code`.

PR #277 (2021-05-07): Remove Git submodules
If installed follow this process to remove local Git submodules:

1. Merge in nasa-jpl:master to which removes the `.gitmodules` file.
2. Delete the following sections from `.git/config`:

```
[submodule "Software/android_app"]
    url = https://github.com/nasa-jpl/osr-android-app
    active = true
[submodule "Software/rover-code"]
    url = https://github.com/nasa-jpl/osr-rover-code
    active = true
```

3. Run `git rm --cached Software/android_app` and `git rm --cached Software/rover-code`.
4. Delete the now untracked submodule files with `rm -r Software/android_app` a [... cut]
commit 68ce41f7 (2021-05-07): Remove Git submodules

If installed follow this process to remove local Git submodules:

1. Merge in nasa-jpl:master to which removes the `.gitmodules` file.
2. Delete the following sections from `.git/config`:

```
[submodule "Software/android_app"]
    url = https://github.com/nasa-jpl/osr-android-app
    active = true
[submodule "Software/rover-code"]
    url = https://github.com/nasa-jpl/osr-rover-code
    active = true
```

3. Run `git rm --cached Software/android_app` and `git rm --cached Software/rover-code`.
4. Delete the now untracked submodule files with `rm -r Software/android_app`  [... cut]
closed 2021-05-07 by Achllle