nasa-jpl/open-source-rover #273: Update BOM/docs with a good alternative for Traxxas 5374X wheels

apollokit, 2021-05-05:
We're encountering out-of-stock's on the Traxxas wheels, for example https://opensourcerover.slack.com/archives/C01D07KNKJM/p1620218389012800

They're currently out of stock on the [dollarhobbyz website link](https://www.dollarhobbyz.com/collections/all/products/traxxas-2-talon-tires-gemini-black-chrome-wheels-5374x) from the BOM, though they can be found [on amazon](https://www.amazon.com/TRAXXAS-5374X-BRUSHLESS-REVO-WHEEL/dp/B00C5T2GRY).

Getting the impression that supply for these wheels is dwindling

Would be good to have a solid alternative for them in the docs/BOM.

1) find an alternative (perhaps taking advantage of the new gobilda motors)
2) update docs/BOM

Achllle, 2021-05-11:
From @dcschooley:

> A question came up somewhere because the default wheels are out of stock. I get mine from Amazon: [Dave's Wheels](https://www.amazon.com/gp/product/B06XS6V7T9/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) They work great. They use a standard 17mm hex mounting. I remove the red ring so that I can remove the factory hubs and replace them with my own design, but the factory hubs could be made to work, I think.

These are the same diameter.

mredmon63, 2021-05-18:
apollokit, are you talking about the Rex 8mm hex output shaft gobilda motors?  I bought two 5203-series motors to evaluate because the BOM motors (5202-series) have been out of stock for six weeks and are not expected for another 5 weeks.
Do you have a recommendation to integrate the Rex 8mm hex output shaft to a wheel?

apollokit, 2021-05-23:
Hmm. Not sure. There's [this hub](https://www.gobilda.com/1309-series-sonic-hub-8mm-rex-bore/) that also has a square pattern for the holes, similar to the [hub we use](https://www.servocity.com/6mm-bore-clamping-d-hub-tapped-0-770-pattern/) to attach to the wheels.

But i don't think the square size is compatible, just from quickly comparing the two. *Do* double check my math on that

Besides that, I would:
- look around for some adapter between the Rex 8mm shaft and the 6mm hub we used...
- or 3d print a custom adapter

@Achllle , thoughts?

I hope this out-of-stock issue is due to current global supply chain issues...

Achllle, 2021-05-24:
> look around for some adapter between the Rex 8mm shaft and the 6mm hub we used...

Would love this scenario, but I just haven't found the parts to do this yet. There seems to be some standardization in the RC cars world, but I've never found documentation for that.

mredmon63, 2021-05-26:
[17mm Hex Adapter.zip](https://github.com/nasa-jpl/open-source-rover/files/6547958/17mm.Hex.Adapter.zip)
This is the simple printed adapter I am using to integrate 17mm wheels to the 7mm hex shaft on the Gobilda motor: https://www.gobilda.com/5203-series-yellow-jacket-planetary-gear-motor-26-9-1-ratio-24mm-length-8mm-rex-shaft-223-rpm-3-3-5v-encoder/
In my opinion this motor is a better option than the BoM motor (5202 series) since it has all-metal gears.  I also prefer the larger diameter hex output shaft over the smaller d-shaft of the 5202 series.  My alternative requires a larger (36mm) clamping mount, M4 0.7mm x 22mm screws (6) and an M4 14mm dia washer (McMaster).  
These are the wheels I chose:
https://www.amazon.com/gp/product/B06XS6V7T9/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
I do not like the looks of them as much as the BoM wheels, but they are comparable.  
I went with the alternate motors and wheels due to lack of availability of the BoM motors and wheels.
Best,
mark

Achllle, 2022-02-07:
@dcschooley just discovered [these Servocity 12mm hex adapters](https://www.servocity.com/hubs-for-12mm-hex-drive-wheels/#actobotics) which might open up the search for new RC wheels without awkward attachment schemes

Achllle, 2023-09-25:
should be resolved as soon as we can get our hands on the Gobilda ones!

closed 2023-09-25 by Achllle