nasa-jpl/open-source-rover #327: Test point designation letter

sethfischer, 2021-07-05:
The designation letter for the control board PCB test points is "T" (T1-T20).

The parts list has references **T1**-**T11** which are spacers and threaded stand-offs.

To avoid ambiguity the test point designator prefix could be changed to "TP", which is recommended by [*ASME Y14.44*](https://www.asme.org/codes-standards/find-codes-standards/y14-44-reference-designations-electrical-electronics-parts-equipment) and [*IEEE Std 315*](https://standards.ieee.org/standard/315-1975.html) *Graphic Symbols for Electrical and Electronics Diagrams (Including Reference Designation Letters)*.

Implementing this change will also improve the relevance of search results when using GitHub search, or searching a local copy of the documentation.

### Notes

1. Other designation letters on the control board PCB are "J" (connectors), "C" (capacitors), and "R" (resistors). All of which comply with the suggestions in section 22.4 of *ASME Y14.44* and *IEEE Std 315*; and do not conflict with a part reference in `parts_list/master_parts_list_raw.csv`.
1. The designation letter "T" is suggested for transformers and coils.
1. Those who don't have access to *ASME Y14.44* or *IEEE Std 315* may refer to https://en.wikipedia.org/wiki/Reference_designator.

sethfischer, 2021-07-12:
It is worth noting that the Arduino shield is using the "TP" designator prefix **TP1**-**TP8** so already complies with *ASME Y14.44*.

Achllle, 2023-09-25:
closing since no longer relevant with latest update

closed 2023-09-25 by Achllle