nasa-jpl/open-source-rover #329: Figure style

sethfischer, 2021-07-05:
Figures are displayed in two styles.

1. Image followed by an emphasized caption:

```markdown
![](../images/calibration/is6.png)
**Figure 6**
```

2. Figure and caption within a table:

```markdown
| <img src="../../images/pcb_assembly/assembly/standoff_5.png" width="300"> |
|:-:|
| Figure 15: Raspberry Pi Standoffs |
```

Standardizing on a single style would improve the consistency of the documentation.

sethfischer, 2021-07-05:
Which is the preferred image style?

ericjunkins, 2021-07-05:
The html version is preferred because it allows you to control properties like width and positioning

Achllle, 2022-01-18:
We want to use as much vanilla markdown as possible, but have been deviating in cases where finer control of the figures is necessary, for example when we're placing two large figures next to one another. My personal take is that it's OK to have both. Requiring HTML everywhere would make the contribution threshold even higher and if the basic markdown works, it works :)

closed 2023-05-13 by Achllle