nasa-jpl/open-source-rover #340: Use relative links for images

sethfischer, 2021-07-17:
## Proposal

Make all image src links relative.

## Justification

Using relative links allows correct rendering of a local preview. For this reason [GitHub's docs recommend using relative links](https://docs.github.com/en/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#relative-links):

> Relative links are easier for users who clone your repository. Absolute links may not work in clones of your repository - we recommend using relative links to refer to other files within your repository.

Being able to preview rendered Markdown before pushing is important as it can reduce the number of errors.

## Example

In `mechanical/head_assembly/README.md` there is one relative link, with the remaining being absolute.

The image with the relative link displays in a local preview with the remaining absolute images being broken.

### Relative link

```markdown
| <img src="../../images/components/structural/S43.png" width="50%"> |
| :--: |
| Figure 1: 3D Printed Parts |
```
[image]

### Absolute link

```markdown
<img src="/images/head/finala.png" width="50%">
```
[image]

sethfischer, 2021-07-21:
If this proposal is accepted I will prepare a pull request.

ericjunkins, 2021-07-21:
Sounds good assigned it to you

PR #345 (2021-08-13): Use relative paths for images
Closes #340
commit 0093cd8f (2021-08-13): Use relative paths for images

Closes #340
closed 2021-08-13 by Achllle