nasa-jpl/open-source-rover #348: Request, Alternative Motor Driver?

BerkayCinar, 2021-11-17:
Roboclaw 2x7A motor drives are no longer manufactured by Pololu. Not available in Turkey.
Is there anything else you can recommend instead of this driver? We are stuck :'( 

[image]
?

Achllle, 2022-01-18:
Hi @BerkayCinar ,
Sorry I missed this! If it's still relevant, I don't believe the RoboClaw's are no longer produced, they're just out of stock with the ship shortages worldwide. You might be able to find them elsewhere, or can use the beefier 2x14A Roboclaws which work the same way.

Achllle, 2023-05-13:
closing due to inactivity

closed 2023-05-13 by Achllle