nasa-jpl/open-source-rover #349: Reference CAD location in Parts List for 3d printed parts

JoshSharpe, 2021-12-21:
Problem: Right now, when looking at the CAD parts that are in various places in the repo, it can be difficult to understand if the part is needed, why it would be, and which part it is on the master parts list. 

This can be solved a few different ways.

- In the parts list, include the directory and file name for the STL file that it refers to
- Include the reference number (ie S24) of the part in the stl file name
- Create a README file for the CAD parts that includes reference number, description of part, location, etc.

Achllle, 2022-01-18:
As a meta-comment, we are considering moving from SolidWorks to OnShape. If you have ideas on how to achieve this with OnShape, if it still could be an issue, I'd love to hear it!

Achllle, 2023-09-25:
closing since no longer relevant with latest update

closed 2023-09-25 by Achllle