nasa-jpl/open-source-rover #354: JST-XH connectors not included in BOM or instructions for GoBilda encoder connections

Achllle, 2022-01-18:
- [ ] add 4 pin JST-XH connectors to BOM
- [ ] add to instructions in right place

JoshSharpe, 2022-01-24:
The GoBilda motor requires these and 3.5mm MH-FC bullet connectors for the M+/- wire that already exist. Usually I'd open a new issue but since these are closely related and should be updated together I wanted to just piggy back off this one.

Depending on instructions, there might be additional parts. I'm planning on using 2.54 mm connector on the other end of the motor to plug into the PCB. I'm not sure if that is what others did. But it seems like that will be more reliable/secure.

Achllle, 2022-01-24:
> The GoBilda motor requires these and 3.5mm MH-FC bullet connectors for the M+/- wire that already exist. Usually I'd open a new issue but since these are closely related and should be updated together I wanted to just piggy back off this one.

Yes, we should probably add these in. They are optional, since you can just solder them together as well without connector.

> Depending on instructions, there might be additional parts. I'm planning on using 2.54 mm connector on the other end of the motor to plug into the PCB. I'm not sure if that is what others did. But it seems like that will be more reliable/secure.

We probably won't make any more updates to existing wiring given the upcoming change to the new PCB 2.0 version. See #pcb-rev-f on slack for the latest updates. Essentially we're replacing the corner motors with servos and the PCB is entirely re-architected. We're aiming for a fall release this year.

Achllle, 2023-09-25:
closing since no longer relevant with latest update

closed 2023-09-25 by Achllle