nasa-jpl/open-source-rover #375: [Bug Report] PCB Assembly Readme.md: Controller Board's JST connector is connected backwards

SallyTYT, 2022-06-28:
The opening in the pins on the JST connector should face OUTWARD as shown below.
[image]

https://github.com/nasa-jpl/open-source-rover/tree/master/electrical/pcb_assembly#362-solder-40-position-headers-2

ericjunkins, 2022-06-28:
Just to clarify, this ticket is to point out that the connector _should have_ been designed to be put on 180 degrees from what it currently is correct? If so yes, that was a design oversight, and in our current work in redesigning a lot of the electrical system, including a new set of PCBs we will make sure to verify these things. 

https://github.com/nasa-jpl/open-source-rover/pull/198

SallyTYT, 2022-06-30:
> Just to clarify, this ticket is to point out that the connector _should have_ been designed to be put on 180 degrees from what it currently is correct? If so yes, that was a design oversight, and in [...]

Yes. 

I found the Arduino shield board couldn't be powered correctly if following the current guidance, then found this issue. The cable E26 given by the BOM list is as the left condition in the description, which means one of the directions is incorrect. I tend to think that generally a connector opening is towards the outside of the PCB, thus, the connector on the control board is the one.

ericjunkins, 2022-06-30:
I am feeling confused still. Is the issue:

1). The directions are incorrect and one of the connectors you think needs to be spun 180 degrees to electrically make it work

OR

2). The board _design_ has two different orientations of connectors and that is awkward because of convention. (I know this is true and being addressed).

SallyTYT, 2022-06-30:
I mean 1) The directions are incorrect and one of the connectors I think needs to be spun 180 degrees to electrically make it work.

We just need to spin the connector itself. I don't think you need to adjust the electrical part of the PCB design. But perhaps you can mark the correct direction on the PCB.

SallyTYT, 2022-07-01:
I painted how the cable direction affects the direction of the connector, as shown in the following figure. The cable we bought is in the right condition.

[image]

> We just need to spin the connector itself. I don't think you need to adjust the electrical part of the PCB design. But perhaps you can mark the correct direction on the PCB.
I'd like to correct that the PCB layout doesn't need to be adjusted, and the schematic and mark may need to be adjusted.

Achllle, 2023-09-25:
closing since no longer relevant with latest update

closed 2023-09-25 by Achllle