nasa-jpl/open-source-rover #412: Inconsistent license

dkogan, 2023-09-23:
Hello.

You have an Apache license that says that you can use and distribute the sources pretty much however you like. And then you also have a disclaimer that places many restrictions on usage and distribution (you have to be a certain age, you need another license to use this commercially, etc, etc). These two statements are incompatible. You should pick which terms you actually want to distribute this under, and then remove the other license statement. I.e. if the limitations stated in the disclaimer are correct, then this isn't licensed under the Apache license, and you should remove your license file.

Thanks

dkogan, 2023-09-23:
I just saw this was addressed in #2, with the official word being that the license file overrides the disclaimer. You should remove the disclaimer. Thanks.

Achllle, 2023-09-24:
Thanks to bringing this to our attention.

> You should

(x3)

Please note that this kind of wording makes it sound like you're telling us what to do and comes over as arrogant. For the last several years, this project has been run by the community and so everyone is a volunteer.

You're welcome to create a PR to remove the extraneous file.

dkogan, 2023-09-24:
Achille Verheye ***@***.***> writes:

> Please note that this kind of wording makes it sound like you're
> telling us what to do and comes over as arrogant. For the last several
> years, this project has been run by the community and so everyone is a
> volunteer.

Hi. Yes. You're 100% right. I apologize, and I definitely do appreciate
your work. Thanks for taking the time to work on this project.

PR #420 (2023-09-26): Remove disclaimer as per issue 412, replace link to license file instead
close #412
closed 2023-09-26 by Achllle