nasa-jpl/open-source-rover #413: Website `opensourcerover.jpl.nasa.gov` is down

maheshrjl, 2023-09-24:
Based on the error, perhaps it is to do with the AWS account permissions. 

[image]

eddmakesio, 2023-09-24:
Yep, similar issue on Safari

```
KMS.UnrecognizedClientExceptionNo account found for the given parametersC87WCZMAYJZ7BX1PKSzn83Z7gWzFJ46RcolBvQSYR97K/w5LGV1rtYSWwNz3P1CkIMKnPsPx8CiQvIXaZ0Uq6zUuJ9A=
```

Achllle, 2023-09-24:
Unfortunately due to spending cuts at JPL, the website is no longer paid for for the time being. I'm hoping it comes back but until then the links will have to be removed. If anyone would like to tackle that, please create a PR

maheshrjl, 2023-09-25:
@Achllle is hosting the page on GitHub pages feasible?

Achllle, 2023-09-25:
I don't know. I'd likely have to get access to the website's source code and I'm not sure if that's worth the effort given that success isn't guaranteed.

maheshrjl, 2023-09-25:
I think we already have everything required for deployment in the repository. I was able to spin up the website using mkdocs on my local.

[image]

[image]

But, deploying to pages would also require modifying the DNS records for `opensourcerover.jpl.nasa.gov`. I see it is currently pointed to a CloudFront distribution.

I can submit a PR for deployment to pages if required. Thanks :)

ericjunkins, 2023-09-25:
The information that is on the github repo and what you have displayed in that mkdocks is not the same as the website that was previously hosted at opensourcerover.jpl.nasa.gov, that was like an educational site about getting to see an animation that was made about the rover, as well as highlighting some of the specifics of how it works.

eddmakesio, 2023-09-25:
Out of interest @ericjunkins / @Achllle is this something that could be crowdfunded/sponsored? Sad to see educational content go offline due to short term funding. Submitted a PR for the link anyway. 

https://github.com/nasa-jpl/open-source-rover/pull/414

Achllle, 2023-09-25:
@mikcox Since @maheshrjl seemingly was confusing the JPL website with our `mkdocs` documentation, I doubt that we can use the source code to host the website ourselves. I think the only way forward is to have JPL reinstate the domain - probably at little or no cost to them.

Achllle, 2024-11-21:
closed by #414, conversation stale

closed 2024-11-21 by Achllle