nasa-jpl/open-source-rover #429: Battery protection circuitry

abust005, 2023-10-23:
There should be some kind of monitoring circuit that cuts off the battery when it reaches a low voltage, to prevent over-discharging LiPos

robrussell, 2024-04-07:
Alarming battery monitors [like these](https://www.amazon.com/Battery-Voltage-Tester-Monitor-Checker/dp/B08832NJ9Z) are popular. It doesn't cut off the power but it does make a really loud alarm when the voltage drops below the set level. It plugs into the balance connector on the LiPo and can be left there while the battery is in use.

This doesn't resolve the issue but could help users in the short-term. Mine didn't come from Amazon, and there might be a better version out there these days.

Achllle, 2024-04-08:
I think this is a good addition, perhaps an optional one. We do have the INA260 and it's worked into the code already so perhaps we can do something with that? Like a relay that can be triggered from the RPi and can be manually switched back on?

TogaTheGreat, 2026-11-02:
Probably just my fault, but I left my battery plugged into the board (switch off) for a few weeks and came back to a 15v battery down to 5v, which I had to get rid of. The only thing I can think of that was actually powered was the Bat+ LED on the motor board, which may have been enough after so much time to kill it. I have a low-voltage monitor now, but it is a bit annoying to have to unplug and put back on every charge- an integrated low-voltage cutoff would be helpful for newbies to LiPos for a future iteration, like a relay or something similar?

Achllle, 2026-11-03:
@TogaTheGreat which board version do you have? This was an issue on v2.0.1 that I believe @apollokit fixed on v2.0.3.

I'd welcome any addition like this. Updating the PCB is not a trivial exercise though, so anything retrofittable would be preferred if possible.