nasa-jpl/open-source-rover #431: Different wire-to-board connection for corner servos

abust005, 2023-10-25:
Current corner servo-to-board connection uses 2.54mm pitch DuPont connectors, which provide no mechanical connection/locking aside from friction and the spring tension of the female connector. The use of **JST-XH** connectors is proposed, as the user already crimps these for the drive motor encoders, and these provide a more stable mechanical connection.

Achllle, 2023-10-25:
Have you experienced problems with these coming undone? In my experience these actually do form a pretty robust connection. I do think switching would be better, but would mean a board revision and documentation update and so likely wouldn't happen in the very near future (unless someone goes for it!)

abust005, 2023-10-25:
Surprisingly, I have had some issues with them coming undone; it's usually my experience that they have a robust connection, but even after several crimpings, I've had some issues with connections coming loose. I won't discount the possibility that my connectors are on the flimsier side though.

On the subject of revving the board and updating the documentation, I can try to tackle it (given that I've been working on that wiring harness diagram I mentioned in the Slack and was thinking of one or two other things); is there anyone in particular I should talk to about board updates?

Achllle, 2023-10-26:
I believe it and don't mind approving of a PR to make that change. @apollokit is the expert on this, though @ericjunkins and @dcschooley are very knowledgeable on this as well. I'd suggest posting in the #osr-upgrade channel in Slack and mentioning them to get their buy-in.

dcschooley, 2024-04-07:
JST connectors would probably be fine. I don't remember how we ended up with the DuPont connectors. It's been a while since I've looked at this.