nasa-jpl/open-source-rover #443: PCB build instructions use part references that are only documented on google sheets

Achllle, 2023-12-10:
Instructions in [pcb build](https://github.com/nasa-jpl/open-source-rover/tree/master/electrical/pcb#33-solder-resistor-r1-and-power-diode-d1) reference things like 'diode D1' and 'resistor R1' but those are nowhere to be found in the repo. They should be specified.

In the meantime, the part references are [here](https://docs.google.com/spreadsheets/d/1Z011-4hdfSKOZROEwKTmGs-w_xfm9FzWOGzqSnaTu4w/edit?usp=sharing). 

Either remove references like D1 and just specify the resistor value + code or specify which is which somewhere.

apollokit, 2024-01-24:
🙈 yeah definitely need to do this

apollokit, 2024-02-05:
Opened a PR here https://github.com/nasa-jpl/open-source-rover/pull/448

apollokit, 2024-04-11:
Done.

closed 2024-04-11 by apollokit