nasa-jpl/open-source-rover #447: Discrepancy in Roboclaw setup vs configuration in config

ericjunkins, 2024-01-14:
The instructions and wiring say that the following should be how it is setup/built:

**RC1**

Address : 128
M1: Front Left
M2: Middle Left

**RC2**

Address: 129
M1: Back Right
M2: Back Left

Address: 130
M1:  Front Right
M2: Middle Right

However that is not what the params file says in the code. 

https://github.com/nasa-jpl/osr-rover-code/blob/master/ROS/osr_bringup/config/roboclaw_params.yaml. 

[image]

Achllle, 2024-01-14:
Can you link to the parts of the instructions and wiring that you're referring to specifically? The [wiring README](https://github.com/nasa-jpl/open-source-rover/blob/master/electrical/wiring/README.md) doesn't deal with roboclaws, it's just instructions for how to make the wires themselves.

ericjunkins, 2024-01-14:
[image]
 I was going with what's on the board, which I have verified is correctly connected to where it says it is on the PCB

Achllle, 2024-01-14:
Woops, looks like I've been pluggin in my cables for the front right motor in the back right and vice versa. Even though I labeled them correctly :1st_place_medal:
Just to verify: the fix you're expecting is to make is to just flip the addresses of `drive_right_back` and `drive_right_front` in `roboclaw_params.yaml`?

ericjunkins, 2024-01-14:
Yes that is my plan to just fix the config to match the PCB

closed 2024-02-16 by Achllle