nasa-jpl/open-source-rover #451: Update standoffs in PCB assembly documentation and BOM

apollokit, 2024-02-09:
Figure out what standoffs to actually use for the PCB, and update the BOM and pcb build readme

Let's go with all metric, to make things easier for the future

Achllle, 2024-02-15:
I'm using:
* 30mm standoffs for the motor board to case connection
*  20mm standoffs for the RPi on the brain board
* 15mm for the Roboclaws
* 10mm for the brain board to motor board connection

I believe these are M2.5 standoffs (the RPi only does M3 if you drill them out). I got some standoff kit on Amazon.

AEmilioDiStefano, 2026-09-19:
Cross-checking the PCB build README against the BOMs at 0bea824 turns up six M2.5 standoff lengths the README calls out and no BOM row for them:

| Length | electrical/pcb/README.md |
|---|---|
| M2.5 x 30 mm | line 52 |
| M2.5 x 20 mm | line 522 |
| M2.5 x 15 mm | line 361 |
| M2.5 x 12 mm + 6 mm (INA260, optional) | line 173 |
| M2.5 x 10 mm | lines 203 (regulator) and 534 (brain to motor board) |

The BOM's only standoff row, B12, states no lengths. Four of the six are the lengths listed above on 2024-02-15; the README adds the 12 mm and 6 mm pair for the INA260.

Reproduce, no tools needed:

    grep -n 'M2.5 x' electrical/pcb/README.md
    grep -in 'standoff' electrical/pcb/control_board/BOM/bom_v2.0.3_other_parts.csv

One thing noticed while checking: README line 203 calls for M2.5 x 10 mm standoffs between the regulator and the motor board, and the U3 footprint (`D24V22Fx_regulator` in `Control_Boards.kicad_pcb`, line 25236) drills its two mounting holes at 2.18 mm, a no. 2 / M2 hole; an M2.5 thread does not pass it.

How many of each length does a build use? With that, I can give you a PR replacing B12 with one row per length.

(The six-length check came from NeuroMux, a consistency checker I'm developing.)