nasa-jpl/open-source-rover #459: Rocker-bogie joint should use bushings instead of bearings

Achllle, 2024-02-24:
There's too much play from the bearings in the rocker-bogie assembly causing the rocker to be a bit flexible which can make the rover look crooked. A bushing would be cheaper and provide a tighter fit. Bushings actually perform better under higher loads when speed are low (which they are)

GoBilda doesn't sell bushings as of now so a solution would involve finding those elsewhere.

Achllle, 2025-04-06:
A few videos illustrating the issue and effect.

https://github.com/user-attachments/assets/e9a6ddd2-7171-4e9e-818a-83102e80f97b

https://github.com/user-attachments/assets/b6fd6e23-8fa4-453c-b6ce-96acecbea84d

Achllle, 2025-04-06:
https://github.com/user-attachments/assets/4d899636-52a9-4a48-8b52-1671a137217c

Achllle, 2025-04-06:
Videos don't seem to work (?)
https://photos.app.goo.gl/qutyniZaA2NoGsm7A
https://photos.app.goo.gl/wWKPRaooJHBW1sgFA

davidgmolnar, 2025-05-25:
Hi, did you find an appropriate bushing?

sKYwALK3R7, 2026-06-26:
> No, GoBilda is not planning on adding bushings to their inventory as far as I'm aware and the preference is to stick with as few suppliers as possible to make the ordering process doable. However the issue is somewhat remediated if the control arms are attached firmly. This is hard to do right with the hurricane nuts but substantially easier with the [goRail 5-hole nut](https://www.gobilda.com/gorail-nut-5-hole-2-pack/).

Did that solve the problem? 
It seems we're having a similar problem. The loose bearing is probably also causing the wheels to be very close together when driving forward (see pictures). 
This problem occurred after a few hours of driving. 

[image]

thvthvhfhb, 2026-09-12:
Since sticking to GoBilda is a constraint, we could potentially use 3D-printed PETG or nylon inserts as a DIY bushing solution to take up the radial clearance currently causing the play. Have we measured the exact inner diameter of the rocker-bogie joint to see if a standard metric oil-impregnated bronze sleeve could be pressed into the existing bore without reaming? If the tolerance stack-up is the main driver of the crooked alignment, replacing the bearing race with a custom-printed sleeve might stabilize the assembly without needing a new supplier.

jasminritik26, 2026-11-04:
To eliminate the play without adding another supplier to the BOM, we could 3D print custom TPU or Igli

brokendenim, 2026-11-14:
If we cannot source external bushings, can we instead implement a shimming solution within the `rockerBogieAssembly` CAD parameters to eliminate the play currently causing the crooked alignment?